intel/intel-vaapi-driver #354: Why H264 Encoder is slower than H265 Encoder?

qiushics, 2018-02-08:
[#184](https://github.com/intel/libva/issues/184)

Does vaapi driver have some difference on SKL?

xhaihao, 2018-02-08:
The driver uses legacy media kernels on HSW. the new kernels are used on BDW+.

xhaihao, 2018-02-08:
Are you using the latest version of intel-vaapi-driver on SKL?

qiushics, 2018-02-08:
@xhaihao 
I used the lastest version of SKL, the performance of H264 increase, but the speed of H265 encoder decrease when compared to the old driver.
I also compare the new version  on both HSW and SKL. HSW has better H264 encoder performance.

The result updated as below:

VAAPI Driver | CPU |Encoder| FPS
------------ | ------------- | ------------- | -------------
1.7.0| i7 4770| H264| 250fps
1.7.0| i7 6700| H264| 108fps
1.7.0 |i7 6700| H265| 126fps
2.1.1.pre1| i7 4770| H264| 250fps
2.1.1.pre1| i7 6700| H264| 200fps
2.1.1.pre1 |i7 6700| H265| 64fps

qiushics, 2018-02-08:
when I use [libva-util](https://github.com/intel/libva-utils)
`sudo intel_gpu_time avcenc 1920 1080 ~/mediatest/1080p.yuv /dev/null`

HSW:
```
encode 683 frames in 3.152886 secondes, FPS is 216.6
user: 0.932000s, sys: 0.400000s, elapsed: 3.154074s, CPU: 42.2%, GPU: 72.6%
```

SKL:
```
encode 683 frames in 3.694241 secondes, FPS is 184.9
user: 2.300000s, sys: 1.132000s, elapsed: 3.697099s, CPU: 92.8%, GPU: 52.6%
```

It seems that SKL can't full utilize the GPU.

xhaihao, 2018-02-09:
> I used the lastest version of SKL, the performance of H264 increased, but the speed of H265 encoder decreased when compared to the old driver.

The quality of the new H265 encoder is far better than the old H265 encoder on SKL. 

> I also compare the new driver on both HSW and SKL. HSW has better H264 encoder performance

A simple media kernel is used on HSW however a more complicated media kernel is used on SKL,  the quality of the h264 encoder on SKL is far better than the h264 encoder on HSW.

> sudo intel_gpu_time avcenc 1920 1080 ~/mediatest/1080p.yuv /dev/null

intel_gpu_time only measures the utilization of the render ring,  actually it is not a good tool for measuring GPU utilization.

qiushics, 2018-02-09:
@xhaihao Thanks.
I know the problems is that the performance is related to the quality encoder.
So, should I use GPA to for measuring GPU cost?

qiushics, 2018-02-13:
ok, thanks

vengateshvengi, 2019-12-12:
As now i am working in ffmpeg with HW acceleration for converstion of mpeg2video to H264 using vaapi in Ubuntu .Converstion is working good but it takes more cpu usage in (wa). I had followed this link
https://gist.github.com/Brainiarc7/24de2edef08866c304080504877239a3
Please guide me to avoid cpu usage in (wa).