intel/intel-vaapi-driver #356: Frame-delay when encoding from a desktop for streaming services

Enigma0, 2018-02-12:
Most notably, Parsec & Steam In-Home-Streaming both exhibit odd frame-delays that make the experience less than smooth.

This is only evident when streaming _to_ a Linux system as far as I am aware. It seems to work ok when it's _to_ a Windows system.

Typing & interacting with anything that relies on visual feedback requires a subsequent refresh action after the initial action for the initial action to show up visually.

xhaihao, 2018-02-13:
Could you attach the output of vainfo?

Enigma0, 2018-02-13:
```
:~$ vainfo
libva info: VA-API version 0.40.1
libva info: va_getDriverName() returns 0
libva info: User requested driver 'i965'
libva info: Trying to open /usr/lib/x86_64-linux-gnu/dri/i965_drv_video.so
libva info: Found init function __vaDriverInit_0_40
libva info: va_openDriver() returns 0
vainfo: VA-API version: 0.40 (libva 1.8.4.pre1)
vainfo: Driver version: Intel i965 driver for Intel(R) Haswell Desktop - 1.8.4.pre1 (1.8.4.pre1)
vainfo: Supported profile and entrypoints
      VAProfileMPEG2Simple            :	VAEntrypointVLD
      VAProfileMPEG2Simple            :	VAEntrypointEncSlice
      VAProfileMPEG2Main              :	VAEntrypointVLD
      VAProfileMPEG2Main              :	VAEntrypointEncSlice
      VAProfileH264ConstrainedBaseline:	VAEntrypointVLD
      VAProfileH264ConstrainedBaseline:	VAEntrypointEncSlice
      VAProfileH264Main               :	VAEntrypointVLD
      VAProfileH264Main               :	VAEntrypointEncSlice
      VAProfileH264High               :	VAEntrypointVLD
      VAProfileH264High               :	VAEntrypointEncSlice
      VAProfileH264MultiviewHigh      :	VAEntrypointVLD
      VAProfileH264MultiviewHigh      :	VAEntrypointEncSlice
      VAProfileH264StereoHigh         :	VAEntrypointVLD
      VAProfileH264StereoHigh         :	VAEntrypointEncSlice
      VAProfileVC1Simple              :	VAEntrypointVLD
      VAProfileVC1Main                :	VAEntrypointVLD
      VAProfileVC1Advanced            :	VAEntrypointVLD
      VAProfileNone                   :	VAEntrypointVideoProc
      VAProfileJPEGBaseline           :	VAEntrypointVLD
```

Enigma0, 2018-02-13:
```
System Information

  PROCESSOR:          Intel Core i7-4790K @ 4.70GHz (8 Cores)
    Core Count:       4
    Thread Count:     8
    Extensions:       SSE 4.2 + AVX2 + AVX + RDRAND + FSGSBASE
    Cache Size:       8192 KB
    Microcode:        0x22
    Scaling Driver:   intel_pstate performance

  GRAPHICS:           Intel Haswell Desktop 1536MB (1250MHz)
    OpenGL:           4.5 Mesa 17.3.0-devel- padoka PPA
    Vulkan:           1.0.57
    Display Driver:   modesetting 1.19.5
    Monitor:          S27C750
    Screen:           1920x1080

  MOTHERBOARD:        ASUS Z97I-PLUS
    Memory:           16384MB
    Chipset:          Intel 4th Gen Core DRAM
    Network:          Intel Connection + Broadcom Limited BCM4352 802.11ac Wireless

  OPERATING SYSTEM:   Ubuntu 17.10
    Kernel:           4.15.0-041500-generic (x86_64)
    Desktop:          GNOME Shell 3.26.2
    Compiler:         GCC 7.2.0
```