intel/intel-vaapi-driver #357: support RGB when exporting a surface to a Wayland buffer

ceyusa, 2018-02-13:
This is a follow up of https://bugzilla.gnome.org/show_bug.cgi?id=775698

The main issue is that mutter, the Wayland compositor for GNOME doesn't support non RGB buffers (see https://gitlab.gnome.org/GNOME/mutter/blob/master/src/wayland/meta-wayland-buffer.c#L232)

The reason why mutter cat import EGL images non-RGB is because cogl, the library that wraps GL for clutter and mutter, doesn't handle YUV textures.

Nonetheless, Weston supports YUV textures (through shaders that convert the frame's color format, as far as I understand).

As exposed in the gstreamer-vaapi bug, it is hard to force all the possible Wayland compositors to support YUV textures, it would be easier to add support the exportation of RGB EGLImages from the Intel VA-API driver (https://github.com/intel/intel-vaapi-driver/blob/master/src/i965_output_wayland.c#L267).

seanvk, 2018-02-23:
@xhaihao Who do you have assigned?

xhaihao, 2018-02-27:
@ceyusa do you mean rgba or rgbx? the driver doesn't support 24bit rgb888 format.

ceyusa, 2018-02-27:
> do you mean rgba or rgbx? 

either, both are handled by Wayland as you can see in the code.

> the driver doesn't support 24bit rgb888 format.

Would be possible use VPP?

seanvk, 2018-03-28:
Any update @jayesht ?