intel/intel-vaapi-driver #363: YUV 4:4:4 JPEG decoding returns garbage for some input files

fhvwy, 2018-02-19:
Sample:
[image]

Decoding appears to succeed, but the output surface contains garbage.  I think the interesting feature of the file here is that V_i is 2 for all components, so the MCU size is actually 1x2.  (This file was created by libavcodec with default settings; YUV 4:4:4 files created by libjpeg have H_i == V_i == 1, and do work.)

yueyihua, 2018-09-26:
When I use yuv444 jpeg data for decoding, the writing has color stain, and sometimes clearly and sometimes indistinctly. Is this the same bug?

yueyihua, 2018-09-26:
@xhaihao，When will this bug be fixed？

yueyihua, 2018-12-03:
@fhvwy Is this bug being fixed?