intel/intel-vaapi-driver #365: Process killed because GPU OOM

YijinLiu, 2018-02-22:
I am using ffmpeg to transcode video using multiple threads.
My process got killed by OOM when the system still has plenty of memory.
After some research, I think it's actually that GPU is short of memory.
According to /sys/kernel/debug/dri/0/i915_gem_objects, almost all memory is in the inactive list.

Some people says it's user space's bug (https://bugs.freedesktop.org/show_bug.cgi?id=94814), though I couldn't find the bug they were mentioning. We are using this driver instead of xf86-video-intel. So I  think if there is some bug, it might be in intel-vaapi-driver.

xhaihao, 2018-03-06:
What's your command line?

xhaihao, 2018-03-06:
> We are using this driver instead of xf86-video-intel.

What do you mean? xf86-video-intel is a 2D driver for X11, not a VA-API backend driver, so you can replace xf86-video-driver with intel-vaapi-driver.

YijinLiu, 2018-03-08:
I have found the culprit. It's partially caused by ffmpeg. ffmpeg has a AVBufferPool that keeps reference to those used hardware frames. Sorry for the false alarm.

closed 2018-03-08 by YijinLiu