intel/intel-vaapi-driver #366: MADI/MCDI produce artifacts on Kabylake

srehm, 2018-02-26:
Using MADI or MCDI I get artifacts that look just like the ones reported here https://bugs.freedesktop.org/show_bug.cgi?id=79765 for SNB and IVB.
I used the same burosch test samples that are linked in that bug report. The result looks identical to https://bugs.freedesktop.org/attachment.cgi?id=100593 (maybe not quite as bad, but that could well be because I only watched it in motion and did not do any screenshots yet). 

Used software:

- intel-vaapi-driver 2.1.0
- mpv
- xbmc

Tested on the following hardware:

- Pentium G4600
- Core  i5 7200u

xhaihao, 2018-03-06:
Is it possible to try this on SKL? I wonder whether it is a KBL specific issue.

srehm, 2018-03-10:
Sorry, I don`t own a SKL CPU. My other Intel machines are all HSW. I can try on one of those if that helps.