intel/intel-vaapi-driver #374: make error. not found va/va_dricommon.h

veranson, 2018-03-14:
CC       libi965_drv_video_la-gen10_vdenc_common.lo
  CC       libi965_drv_video_la-gen10_vdenc_vp9.lo
  CC       libi965_drv_video_la-i965_output_dri.lo
i965_output_dri.c:27:29: fatal error: va/va_dricommon.h: No such file or directory
 #include <va/va_dricommon.h>
                             ^
compilation terminated.
make[3]: *** [libi965_drv_video_la-i965_output_dri.lo]

xhaihao, 2018-03-14:
Did you install libva-drm? You may run 'pkg-config --libs libva-drm'  check libva-drm is installed or not.

veranson, 2018-03-15:
build and install
[root@localhost src]# pkg-config --libs libva-drm
-L/root/apps/libva/lib -lva-drm -lva  
[root@localhost src]#

j-santander, 2019-02-25:
Had the same problem, it turned out that my build of libva did not include X11 support.

topilski, 2019-12-09:
Same issue

iliuhe, 2020-07-10:
I solved this issue by recompile my libva. When compiling the libva, I edit the configure.ac file
1. change this to yes enable_x11="yes"
```
AC_ARG_ENABLE(x11,
    [AC_HELP_STRING([--enable-x11],
                    [build with VA/X11 API support @<:@default=auto@:>@])],
    [], [enable_x11="yes"])
```
2. change this use_x11 to yes
```
# Check for X11
USE_X11="yes"
```
then recompile libva
```
./autogen.sh
./configure
make
make install
cp /usr/local/include/va/* /usr/include/ 
cp /usr/local/lib/libva* /usr/lib64/
```
Finally recompile this and it should work