intel/intel-vaapi-driver #383: [LIBVA] Trying to use iGPU quicksync when Nvidia card is used as main display

ICHx, 2018-05-01:
Dear devs,
I have a PC with the option to empower both PCIe and iGPU graphics at the same time, however, I could not test quicksync ability using vainfo.  Perhaps I need some direction here?

https://forum.manjaro.org/t/libva-trying-to-use-igpu-quicksync-when-nvidia-card-is-used-as-main-display/43809/16

When I try  vainfo:
```
[devops@devops-pc ~]$ LIBVA_DRIVER_NAME=i965 vainfo
libva info: VA-API version 1.1.0
libva info: va_getDriverName() returns 0
libva info: User requested driver ‘i965’
libva info: Trying to open /usr/lib/dri/i965_drv_video.so
libva info: Found init function __vaDriverInit_1_1
libva error: /usr/lib/dri/i965_drv_video.so init failed
libva info: va_openDriver() returns -1
vaInitialize failed with error code -1 (unknown libva error),exit
```

When I try ffmpeg, it did generate a video file
```
[devops@GlaDOS Movies]$ LD_PRELOAD=/usr/lib/dri/i965_drv_video.so LIBVA_DRIVER_NAME=i965 ffmpeg -hwaccel vaapi -vaapi_device /dev/dri/renderD129 -i '2018-04-29 21-36-03.mkv' -vf 'format=nv12|vaapi,hwupload' -threads 8 -c:v h264_vaapi -preset slower -c:a libopus "f-new.mkv"   
ffmpeg version 3.4.2 Copyright (c) 2000-2018 the FFmpeg developers
  built with gcc 7.3.0 (GCC)
  configuration: --prefix=/usr --disable-debug --disable-static --disable-stripping --enable-avisynth --enable-avresample --enable-fontconfig --enable-gmp --enable-gnutls --enable-gpl --enable-ladspa --enable-libass --enable-libbluray --enable-libfreetype --enable-libfribidi --enable-libgsm --enable-libiec61883 --enable-libmodplug --enable-libmp3lame --enable-libopencore_amrnb --enable-libopencore_amrwb --enable-libopenjpeg --enable-libopus --enable-libpulse --enable-libsoxr --enable-libspeex --enable-libssh --enable-libtheora --enable-libv4l2 --enable-libvidstab --enable-libvorbis --enable-libvpx --enable-libwebp --enable-libx264 --enable-libx265 --enable-libxcb --enable-libxml2 --enable-libxvid --enable-shared --enable-version3 --enable-omx
  libavutil      55. 78.100 / 55. 78.100
  libavcodec     57.107.100 / 57.107.100
  libavformat    57. 83.100 / 57. 83.100
  libavdevice    57. 10.100 / 57. 10.100
  libavfilter     6.107.100 /  6.107.100
  libavresample   3.  7.  0 /  3.  7.  0
  libswscale      4.  8.100 /  4.  8.100
  libswresample   2.  9.100 /  2.  9.100
  libpostproc    54.  7.100 / 54.  7.100
Input #0, matroska,webm, from '2018-04-29 21-36-03.mkv':
  Metadata:
    ENCODER         : Lavf57.83.100
  Duration: 00:00:17.10, start: 0.000000, bitrate: 1373 kb/s
    Stream #0:0: Video: h264 (Constrained Baseline), yuv420p(progressive), 1280x720, 1k fps, 30 tbr, 1k tbn, 60 tbc (default)
    Metadata:
      DURATION        : 00:00:17.100000000
    Stream #0:1: Audio: aac (LC), 44100 Hz, stereo, fltp (default)
    Metadata:
      title           : Track1
      DURATION        : 00:00:17.090000000
Codec AVOption preset (Configuration preset) specified for output file #0 (f-new.mkv) has not been used for any stream. The most likely reason is either wrong type (e.g. a video option with no video streams) or that it is a private option of some encoder which was not actually used for any stream.        
File 'f-new.mkv' already exists. Overwrite ? [y/N] y
Stream mapping:
  Stream #0:0 -> #0:0 (h264 (native) -> h264 (h264_vaapi))
  Stream #0:1 -> #0:1 (aac (native) -> opus (libopus))
Press [q] to stop, [?] for help
[libopus @ 0x558643eba880] No bit rate set. Defaulting to 96000 bps.
Output #0, matroska, to 'f-new.mkv':
  Metadata:
    encoder         : Lavf57.83.100
    Stream #0:0: Video: h264 (h264_vaapi) (High) (H264 / 0x34363248), vaapi_vld, 1280x720, q=0-31, 30 fps, 1k tbn, 30 tbc (default)
    Metadata:
      DURATION        : 00:00:17.100000000
      encoder         : Lavc57.107.100 h264_vaapi
    Stream #0:1: Audio: opus (libopus) ([255][255][255][255] / 0xFFFFFFFF), 48000 Hz, stereo, flt, 96 kb/s (default)
    Metadata:
      title           : Track1
      DURATION        : 00:00:17.090000000
      encoder         : Lavc57.107.100 libopus
Invalid return value 0 for stream protocol time=00:00:16.59 bitrate=1001.9kbits/s speed=6.62x    
    Last message repeated 1 times
frame=  513 fps=197 q=-0.0 Lsize=    2222kB time=00:00:17.11 bitrate=1063.4kbits/s speed=6.58x    
video:2209kB audio:3kB subtitle:0kB other streams:0kB global headers:0kB muxing overhead: 0.439519%

```

ICHx, 2018-05-01:
```
Spec:
System:    Host: GlaDOS Kernel: 4.16.6-1-MANJARO x86_64 bits: 64 compiler: gcc v: 7.3.1
           Desktop: KDE Plasma 5.12.4 tk: Qt 5.10.1 Distro: Manjaro Linux 17.1.9 Hakoila
Machine:   Type: Desktop System: ASUS product: All Series v: N/A serial: N/A
           Mobo: ASUSTeK model: H81M-E v: Rev X.0x serial: N/A UEFI: American Megatrends v: 3601
           date: 12/15/2017
CPU:       Dual Core: Intel Celeron G1840 type: MCP arch: Haswell speed: 800 MHz min/max: 800/2800 MHz
Graphics:  Card-1: Intel Xeon E3-1200 v3/4th Gen Core Processor Integrated Graphics driver: i915 v: kernel
           bus ID: 00:02.0
           Card-2: NVIDIA GP108 [GeForce GT 1030] driver: nvidia v: 396.18 bus ID: 01:00.0
           Display: x11 server: X.Org 1.19.6 driver: modesetting,nvidia resolution: 1920x1080~60Hz
           OpenGL: renderer: GeForce GT 1030/PCIe/SSE2 v: 4.6.0 NVIDIA 396.18 direct render: Yes
Network:   Card-1: Realtek RTL8111/8168/8411 PCI Express Gigabit Ethernet driver: r8168 v: 8.045.08-NAPI
           port: d000 bus ID: 03:00.0
Drives:    HDD Total Size: 698.65 GiB used: 210.04 GiB (30.1%)
Info:      Processes: 180 Uptime: 45m Memory: 7.67 GiB used: 2.60 GiB (33.9%) Init: systemd Compilers:
           gcc: 7.3.1 Shell: bash v: 4.4.19 inxi: 3.0.07
```

xhaihao, 2018-05-04:
Sorry, I forgot drm is required. The right command should be:

$> LIBVA_DRIVER_NAME=i965 vainfo --display drm --device /dev/dri/renderD129

closed 2018-05-05 by ICHx