intel/intel-vaapi-driver #397: GPU Hang/System Crash when VPP Sharpen on 1280x720 input

uartie, 2018-06-01:
When doing vpp sharpen on 1280x720 input stream, the GPU hangs and sometimes crashes the system.

```
[drm] GPU HANG: ecode 9:0:0x8fd0ffff, in yamivpp [30257], reason: No progress on rcs0, action: reset
i915 0000:00:02.0: Resetting rcs0 after gpu hang
```

I tested this on kernels >= 4.15, various platforms >= BDW and with yamivpp and gstreamer and every environment encounters this issue.

The iHD driver does not exhibit this problem... only i965.  Also, other resolutions (e.g. 1920x1080) seem fine.

```
$ gst-launch-1.0 videotestsrc num-buffers=10 \
  ! video/x-raw,format=NV12,width=1280,height=720 \
  ! filesink location=1280x720.NV12
$ yamivpp -s 20 1280x720.NV12 1280x720_yamisharp.NV12
$ gst-launch-1.0 -vf filesrc location=./1280x720.NV12 \
  ! rawvideoparse format=nv12 width=1280 height=720 \
  ! vaapipostproc sharpen=0.5 \
  ! filesink location=1280x720_gstsharp.NV12
```

uartie, 2019-03-08:
GPU HANG occurs with 4k content, too.

uartie, 2019-03-08:
Can also reproduce with ffmpeg + i965

zhoub, 2019-06-12:
Bump, also have this issue.