intel/intel-vaapi-driver #398: Scaling from P010 to NV12 has bad output on Apollo Lake

fhvwy, 2018-06-05:
Scaling from P010 to NV12 seems to do something nasty to the chroma plane of the output.

For example:
```
ffmpeg -y -hwaccel vaapi -hwaccel_output_format vaapi -i in.mp4 -an -vf 'scale_vaapi=w=1278:h=720:format=nv12' -c:v h264_vaapi out.mp4
```
Input and output files: [p010_nv12.zip](https://github.com/intel/intel-vaapi-driver/files/2073670/p010_nv12.zip).

It doesn't happen when not scaling (direct conversion P010 to NV12), but does happen at any scale factor in either dimension for both upscaling and downscaling.

The input and output method doesn't seem to matter - with decoding and/or encoding done in software (with corresponding upload and/or download operations) the output is the same.

Seen on both J3455 and N4200.  Not reproducible on other tested gen9 platforms - works on Coffee Lake, not possible on Skylake due to lack of P010 support.

aaronwolen, 2018-08-10:
Good additional information about this issue [here](https://forums.plex.tv/t/color-corruption-on-resize-when-hw-transcoding-hevc-content-to-h264-4k-subs-synology-ds918-nas/223645/5)

KamistixX, 2018-09-26:
J3355 is also impacted.

ValZapod, 2021-02-27:
> Good additional information about this issue [here](https://forums.plex.tv/t/color-corruption-on-resize-when-hw-transcoding-hevc-content-to-h264-4k-subs-synology-ds918-nas/223645/5) with a simple workaround.

Removing :format=nv12 is not a fix. At all.

ValZapod, 2021-02-27:
Can this fix it? https://patchwork.ffmpeg.org/project/ffmpeg/patch/qv2PyTKpWVd6zJqsf3pwRO7gzci3X9OW3r2v3Kt9z1k@cp4-web-037.plabs.ch/