intel/intel-vaapi-driver #403: VPP: the output will always same when set different denoise level.

feiwan1, 2018-07-05:
By using libyami/gstreamer-vaapi, and set different denoise value, the output always bitmatch.

gstreamer-vaapi issue link:
https://bugzilla.gnome.org/show_bug.cgi?id=796635

libyami issue link:
https://github.com/intel/libyami/issues/861

And we use gstreamer-vaapi test on iHd drvier, it looks good.

uartie, 2018-10-18:
same issue is observed with ffmpeg-vaapi, too.

dseeker, 2019-01-25:
seems like the same issue I'm finding. I'm using ffmpeg with a [docker image](https://hub.docker.com/r/jrottenberg/ffmpeg/), here's the [commit ](https://git.arkena.net/ngaullie/ffmpeg/commit/9bba10c174c893b08b036898a522ed6dad1d3660)to the implementation

I messaged @mypopydev about this issue. I noticed if I go over the specified limits it gives out a warning, so it tells me the feature is active in this build, however at any setting up until 64  there is no change in the image or the output file size

uartie, 2019-06-14:
@xhaihao can you take a look at this issue?

srware, 2019-07-24:
Bumping this issue. I have also noticed that the denoise filter doesn't seem to have any effect with this driver on multiple platforms with Gen9 graphics (maybe others).