intel/intel-vaapi-driver #419: use wl_drm_interface from libva and not libEGL

tripzero, 2018-10-17:
intel-vaapi-driver tries to dlopen libEGL.so.1 and find the wl_drm_interface symbol from it.  As of libva/c26a187cdff82dc5a360676ef05e6f6d761c2a0a libva provides its own internal implementation.  Shouldn't intel-vaapi-driver do the same?

jwang11, 2018-12-03:
Is there anybody working on this issue? It block the basic playback function on latest Mesa (libEGL)

xhaihao, 2018-12-04:
@jwang11 @tripzero #419 removes the dependency on libEGL, but I just did limited tests on Wayland. Could you give a try with this patch?

jwang11, 2018-12-06:
@xhaihao , the patch fix the problem. I tested with both yami and gst, both ok now.
- Gstreamer to play h264
```
GST_GL_PLATFORM=egl gst-launch-1.0 filesrc location=media.h264 ! h264parse ! vaapih264dec ! vaapipostproc ! glimagesink
```

- yami to play h264
```
$tests/yamidecode -i media.h264 –m 5
```

xhaihao, 2018-12-07:
The driver has explicit dependency on libwayland-client.so after applying https://github.com/xhaihao/intel-vaapi-driver/commit/654a9e035b4a43e400dc56da0be786a83886aa76, which is not what we want. 

wl_drm_interface in libva (actually in libva-wayland) is not exported too. We should do some changes in libva if we want to use wl_drm_interface from libva.

jwang11, 2018-12-10:
The patches are workable in ClearLinux with Wayland backend.

closed 2019-01-11 by xhaihao

xry111, 2019-08-26:
Will we make a release containing this fix?