intel/intel-vaapi-driver #421: GPU hang when MJPEG decoding with VAAPI in CPU Intel(R) Celeron(R) CPU J1900 @ 1.99GHz

yueyihua, 2018-10-22:
[  409.956007] [drm] GPU HANG: ecode 7:0:0x85fdfffd, in decode_thread [6694], reason: Hang on render ring, action: reset
[  409.956011] [drm] GPU hangs can indicate a bug anywhere in the entire gfx stack, including userspace.
[  409.956013] [drm] Please file a _new_ bug report on bugs.freedesktop.org against DRI -> DRM/Intel
[  409.956014] [drm] drm/i915 developers can then reassign to the right component if it's not a kernel issue.
[  409.956015] [drm] The gpu crash dump is required to analyze gpu hangs, so please always attach it.
[  409.956017] [drm] GPU crash dump saved to /sys/class/drm/card0/error
[  409.956142] drm/i915: Resetting chip after gpu hang
[  419.940003] drm/i915: Resetting chip after gpu hang

[GPU-dump-error.log](https://github.com/intel/intel-vaapi-driver/files/2500679/GPU-dump-error.log)

yueyihua, 2018-10-22:
I Just using vaapi decoding, and copy it to CPU, but not do rendering directly.

zhoub, 2019-06-04:
Bump.

@yueyihua  Do you have this issue  anymore ?

yueyihua, 2019-06-04:
Sorry, I am not care about this bug now.------------------ Original ------------------

 
 
&mdash;