intel/intel-vaapi-driver #422: [SNB] GPU HANG: ecode 6:2:0xec869ffe, in mpv/vo [2923], reason: hang on vcs0, action: reset

ValdikSS, 2018-10-24:
The following video file instantly crashes Intel HD 3000 (Sandy Bridge) GPU, when decoded with mpv/VAAPI.
[gpucrash.zip](https://github.com/intel/intel-vaapi-driver/files/2511733/gpucrash.zip)

Tested on Fedora 4.18.13-200.fc28.x86_64, with libva-intel-driver-2.1.0-2.fc28.x86_64 and libva-2.1.0-1.fc28.x86_64.

Here are the logs with `drm.debug=0x1e log_buf_len=4M`, with a bit older libva version and drm-tip kernel.
[gpucrash_drmtip_afd14c4.zip](https://github.com/intel/intel-vaapi-driver/files/2511747/gpucrash_drmtip_afd14c4.zip)

Kernel developers said it's libva bug: https://bugs.freedesktop.org/show_bug.cgi?id=106676

kwizart, 2019-03-04:
I don't reproduce any crash with your file on Intel Atom Z36xxx/Z37xxx Series Graphics with fedora 29 x86_64 with libva-2.4.0 and libva-intel-driver 2.3.0
Can you reproduce on f29 ?

(I'm experiencing few stability issue thought, debug in progress)...

ValdikSS, 2019-03-04:
libva-2.4.0-2.fc29.x86_64, libva-intel-driver-2.3.0-2.fc29.x86_64
Now it doesn't crash but hangs for about 15-20 seconds, then produces green video.
I was trying to debug this issue with @kagami, to no avail. We figured out that this video is encoded in slices mode and contains 8 slices. HD3000 decodes only odd slices, procuding half of vertical resolution of (every) single frame. One might think that the problem is with even slices, BUT replacing odd slice with even slice manually using hex editor successfully decodes the slice.