intel/intel-vaapi-driver #424: Assertion failed when run vainfo in Intel(R) Core(TM) i3-7100U CPU @ 2.40GHz

yueyihua, 2018-11-29:
libva info: VA-API version 0.40.0
libva info: va_getDriverName() returns 0
libva info: Trying to open /usr/lib64/dri/i965_drv_video.so
libva info: Found init function __vaDriverInit_0_40
vainfo: intel_memman.c:38: intel_memman_init: Assertion `intel->bufmgr' failed.

Program received signal SIGABRT, Aborted.
0x00007ffff64a9a98 in raise () from /lib64/libc.so.6
(gdb) bt
#0  0x00007ffff64a9a98 in raise () from /lib64/libc.so.6
#1  0x00007ffff64ab69a in abort () from /lib64/libc.so.6
#2  0x00007ffff64a2227 in __assert_fail_base () from /lib64/libc.so.6
#3  0x00007ffff64a22d2 in __assert_fail () from /lib64/libc.so.6
#4  0x00007ffff51a337a in intel_memman_init (intel=intel@entry=0x55555575ac10) at intel_memman.c:38
#5  0x00007ffff51a3020 in intel_driver_init (ctx=0x555555759070) at intel_driver.c:126
#6  0x00007ffff517228d in i965_Init (ctx=0x555555759070) at i965_drv_video.c:6894
#7  __vaDriverInit_0_40 (ctx=0x555555759070) at i965_drv_video.c:7056
#8  0x00007ffff7bbf26e in va_openDriver (dpy=dpy@entry=0x555555759260, driver_name=driver_name@entry=0x555555759740 "i965") at va.c:383
#9  0x00007ffff7bc005b in vaInitialize (dpy=0x555555759260, major_version=0x7fffffffe3a0, minor_version=0x7fffffffe3a4) at va.c:647
#10 0x0000555555555924 in main ()

xhaihao, 2018-12-04:
Please update  libdrm, libva and i965 driver.

yueyihua, 2018-12-04:
@xhaihao Which version should I use? 
My version is:
libdrm-2.4.65-1.fc23.x86_64
libva-1.8.3-1.fc23.x86_64
libva-intel-driver-1.8.3-1.fc23.x86_64

xhaihao, 2018-12-11:
Could you provide the output of 'lspci -nn' ?

yueyihua, 2018-12-14:
@xhaihao Oh, sorry, I have changed another device but not before i3-7100U CPU.