intel/intel-vaapi-driver #425: i915/kbl: GPU HANG: ecode 9:2:0xa8dfbffd, in VideoPlayer [770], reason: hang on vcs0, action: reset

mlauss, 2018-12-04:
2160p HEVC video, played on a 1080 output.  One frame of the video is displayed, then the vcs0 hang occurs.  using latest -git versions of libva and intel-vaapi-driver.
Attached the gpu crash dump.
[kblcrash.txt](https://github.com/intel/intel-vaapi-driver/files/2644024/kblcrash.txt)

xhaihao, 2018-12-06:
Which video player are you running? if possible, could you share your video sample?

mlauss, 2018-12-06:
Video Player is Kodi (XBMC), latest git snapshot.  I've extracted a 10 second sample (~64MB), do you have a location I can upload it to?

xhaihao, 2018-12-11:
Are you able to reproduce this issue with other players, such as mpv? Sorry I haven't a location for uploading. If possible, you may upload your video sample to a free cloud storage, then share the link.

mlauss, 2018-12-12:
Hm, I've tried with a recent git snapshot of mpv,  "mpv --hwdec=vaapi --vo=vaapi brokenfile.mkv" and it plays it just fine.   I'm closing this one, is probably an issue with the player in kodi/xbmc.

closed 2018-12-12 by mlauss