intel/intel-vaapi-driver #436: Failed to resolve wl_drm_interface(): /usr/lib/libEGL_mesa.so.0

Kibouo, 2019-01-10:
Using X11 and get following error when running `vainfo`. I think it has something to do with #419 ? That talks about a fix for Wayland being pushed?
```
error: failed to resolve wl_drm_interface(): /usr/lib/libEGL_mesa.so.0: undefined symbol: wl_drm_interface
vaInitialize failed with error code -1 (unknown libva error),exit
```
-  libva: 2.3.0-1
-  libva-intel-driver: 2.3.0-1
-  mesa: 18.3.1-1
-  libx11: 1.6.7-1
(If I'm missing some info please say so.)

xhaihao, 2019-01-11:
A duplicate of #419

closed 2019-01-11 by xhaihao

Kibouo, 2019-01-11:
Sincere apologies if I'm missing something, but how is this a duplicate? Fix #427 for #419 only addresses Wayland.

I am not using Wayland at all.