intel/intel-vaapi-driver #444: vaapijpegdec fails to decode Logitech C930e jpeg image

ChingMLai, 2019-02-09:
We use “vaapijpegdec” with “vaapipostproc” to decode video in stream of jpeg images at frame size of 1280 x 720 from a number of webcams.  It performs well for Logitech C500, C525, C615, C920, C922, as well as Microsoft LifeCam HD1080.  However, it fails to decompress video of Logitech C930e.  It may produce one or two frames, but then the pipeline stalls.  The decoded first two frames are not right either, as it misses the color and contains ghost images.  For your information, the “guvcview” software decoder can decompress the jpeg images of C930e, as well as those from all other webcams, without any problem.  It appears C930e contains extra tag that is rejected by “vaapijpegdec” and it causes the pipeline to halt.   We also test Gstreamer software decoder “jpegdec” with “videoconvert”.  It fails on C930e also, although it passes all other webcams.  For C930e, “jpegdec”  produces wrong images similar to that of “vaapijpegdec” and Gstreamer prints out many error messages.

zhoub, 2019-04-30:
Did you check the camera pixel format by v4l2-ctl ? Usually the cameras has different type of pixel formats.

ChingMLai, 2019-04-30:
///  2019-04-30
Hi Bo Zhou,

	Thank you for your attention to this issue.

	Yes, I use  v4l2-ctl to select the format as   V4L2_PIX_FMT_MJPEG.

	By the statements of -
 	v4l2_format fmt;
	fmt.fmt.pix.pixelformat = V4L2_PIX_FMT_MJPEG;

	There are three points worth noting,

	(1)  The function vaapijpegdec (as well as jpegdec) works fine for Logitech C500, C525, C615, C920, C922, as well as Microsoft LifeCam HD1080.  It fails for C930e, and maybe some other newer models.  I tested them just swapping the webcams on the USB port.

	(2)  All the webcams work fine in the “guvcview” tool.  They all work fine also when I switch the captured jpeg data to feed a jpeg-decoder software function (as used in the guvcview tool). It indicates the jpeg data captured from C930e is normal and good as other models.

	(3)  Sometime I get the first frame of C930e OK.  And sometime I even get the second frame too.  But then I cannot continue to get the rest of video frames.  It indicates that vaapijpegdec (as well as jpegdec) works to some extend for C930e.  But then, it may not be able to handle certain tags in C930e and fail to continue.

	Best regards,

Ching-Ming Lai