intel/intel-vaapi-driver #448: gen9_hcpd_get_reference_picture_frame_id: Assertion `0' failed.

gniezen, 2019-03-11:
When attempting to play a video in the default Ubuntu video player Totem, the application crashes with the following assertion failure whenever I try to seek to a different position in the video:

```
totem: gen9_mfd.c:649: gen9_hcpd_get_reference_picture_frame_id: Assertion `0' failed.
```

I first reported it as an [Ubuntu bug](https://bugs.launchpad.net/ubuntu/+source/intel-vaapi-driver/+bug/1818862), but it was suggested that I report it here as the issue seems to be with `intel-vaapi-driver`. Backtrace/debug info is available at the original bug report on Ubuntu.

compiaffe, 2019-04-15:
A similar issue happens for us when using vaapi in GStreamer. Corresponding bug report is here: https://gitlab.freedesktop.org/gstreamer/gstreamer-vaapi/issues/98

mistralol, 2019-04-25:
Still seeing this in the gstreamer environment in ubuntu 18.04. Some more information. It happens only when the stream is started on a non key frame. So from what I can tell the decoder has no reference frame and crashes when given the first delta.

Some debug info fromt he core file that was generated.

`#4  0x00007f6161260b41 in gen9_hcpd_get_reference_picture_frame_id (frame_store=frame_store@entry=0x7f615c121280, ref_pic=<optimised out>, ref_pic=<optimised out>) at gen9_mfd.c:649
#5  0x00007f616126249b in gen9_hcpd_get_collocated_ref_idx (ctx=0x7f615c068e30, gen9_hcpd_context=0x7f615c121250, slice_param=0x7f615c078820, pic_param=0x7f615c09ac00) at gen9_mfd.c:806
#6  0x00007f616126249b in gen9_hcpd_slice_state (ctx=0x7f615c068e30, gen9_hcpd_context=0x7f615c121250, next_slice_param=0x7f615c09c100, slice_param=0x7f615c078820, pic_param=0x7f615c09ac00)
    at gen9_mfd.c:878
`

`
(gdb) p *frame_store
$2 = {
  surface_id = 4294967295, 
  frame_store_id = -1, 
  obj_surface = 0x0, 
  ref_age = 0
}
`

`
(gdb) p frame_store[0]
$3 = {
  surface_id = 4294967295, 
  frame_store_id = -1, 
  obj_surface = 0x0, 
  ref_age = 0
}
(gdb) p frame_store[1]
$4 = {
  surface_id = 4294967295, 
  frame_store_id = -1, 
  obj_surface = 0x0, 
  ref_age = 0
}
(gdb) p frame_store[2]
$5 = {
  surface_id = 4294967295, 
  frame_store_id = -1, 
  obj_surface = 0x0, 
  ref_age = 0
}
(gdb) p frame_store[3]
$6 = {
  surface_id = 4294967295, 
  frame_store_id = -1, 
  obj_surface = 0x0, 
  ref_age = 0
}
(gdb) p frame_store[4]
$7 = {
  surface_id = 4294967295, 
  frame_store_id = -1, 
  obj_surface = 0x0, 
  ref_age = 0
}
`

`
#5  0x00007f616126249b in gen9_hcpd_get_collocated_ref_idx (ctx=0x7f615c068e30, gen9_hcpd_context=0x7f615c121250, slice_param=0x7f615c078820, pic_param=0x7f615c09ac00) at gen9_mfd.c:806
806	in gen9_mfd.c
(gdb) p pic_param
$8 = (VAPictureParameterBufferHEVC *) 0x7f615c09ac00
(gdb) p *pic_param
$9 = {
  CurrPic = {
    picture_id = 67108864, 
    pic_order_cnt = 10, 
    flags = 0, 
    va_reserved = {0, 0, 0, 0}
[... 238 lines cut ...]

`
#7  gen9_hcpd_hevc_decode_picture (ctx=0x7f615c068e30, decode_state=0x7f615c06c858, gen9_hcpd_context=0x7f615c121250) at gen9_mfd.c:1027
1027	in gen9_mfd.c
(gdb) p *decode_state 
$15 = {
  base = {
    chroma_formats = 1
  }, 
  pic_param = 0x7f615c0791f0, 
  slice_params = 0x7f615c079410, 
  iq_matrix = 0x0, 
  bit_plane = 0x0, 
  huffman_table = 0x0, 
  slice_datas = 0x7f615c078770, 
  probability_data = 0x0, 
  current_render_target = 67108864, 
  max_slice_params = 10, 
  max_slice_datas = 10, 
  num_slice_params = 3, 
  num_slice_datas = 3, 
  render_object = 0x7f615c072b90, 
  reference_objects = {0x0 <repeats 16 times>}
}
`

Climax777, 2019-10-30:
Any workarounds?

mistralol, 2019-10-30:
For my case there was. Which was in a gstreamer pipeline. Put a gstreamer element in front which drops all frames until a key frame is seen.

If a caps negotiation happens. reset the state to drop frames until a key frame is seen again.

Climax777, 2019-10-30:
A custom element I'm assuming?

mistralol, 2019-10-30:
Yes. They are not hard to write for something that simple.
Example: https://github.com/mistralol/gstreamer-js/blob/master/src/dropdeltas.c

Climax777, 2019-10-30:
Thanks I see how it will work for dropping deltas. That specific
implementation doesn't have the mechanism to stop dropping, or am I missing
something?

gniezen, 2019-10-31:
FWIW, this seems to be fixed in Totem (https://bugs.launchpad.net/ubuntu/+source/totem/+bug/1821525) but I think the solution was just to remove the VAAPI plugin.

ankit22july, 2019-12-10:
I am also getting same i965 driver assertion failed error while seeking to IRAP nal unit( other than IDR), as for iHD driver code is working fine. For my use-case i am getting CRA nal unit on seeking as the file only contains one IDR at starting.

compiaffe, 2020-02-26:
In GStreamer (which partly?! powers totem) this has been fixed and backported 2 month ago: https://gitlab.freedesktop.org/gstreamer/gstreamer-vaapi/-/merge_requests/199

jlucke86, 2020-04-21:
@mistralol @Climax777 

I built and tested the dropdeltas.c custom pipeline element and inserted it into my pipeline. Some of the gstreamer work I'm doing lately is on slightly lossy point to point radios (1000+ meters). Its natural that another keyframe is lost and the intel vaapi just crashes a few seconds later.

Are you guys or anyone else for that matter aware of a real fix?

mistralol, 2020-04-21:
The example I posted isn't complete. You need to change the logic in dropdeltas so that it stops dropping deltas when a key frame is seen.

WMichelsen, 2020-08-28:
@jlucke86 
Your use case appears similar to mine, gstreamer vaapi from an imperfect radio link resulting in this ref_pic assertion fail.
Did you modify @mistralol 's dropdelta code into a more robust element as mentioned?  If so, do you mind sharing?  I'm willing to do so, but I'm new to gstreamer and looking for the best starting point.  And, thanks @mistralol .
Also hoping a permanent fix happens.  In my use case, this is a problem with h265parse.  h264parse seems to be tolerant of this problem.