intel/intel-vaapi-driver #451: totem assert failure: totem: totem: gen9_mfd.c:649: gen9_hcpd_get_reference_picture_frame_id: Assertion `0' failed.

ejsc1, 2019-03-24:
Open bug in launchpad.net https://bugs.launchpad.net/bugs/1821525

"Watching movie 4k."

BackTrace:
```
#0  __GI_raise (sig=sig@entry=6) at ../sysdeps/unix/sysv/linux/raise.c:50
        set = {__val = {0, 139667910656424, 4222451712, 139666431533520, 139666431533520, 139666431533520, 139666431533520, 139666431533619, 139666431533620, 139666431533520, 139666431533620, 0, 0, 0, 0, 0}}
        pid = <optimized out>
        tid = <optimized out>
        ret = <optimized out>
#1  0x00007f06f82d2535 in __GI_abort () at abort.c:79
        save_stage = 1
        act = {__sigaction_handler = {sa_handler = 0x7f06a009e1d0, sa_sigaction = 0x7f06a009e1d0}, sa_mask = {__val = {0, 139666430885920, 4294967295, 0, 0, 0, 21474836480, 139666758029096, 139666758028944, 139667817901418, 139667817901418, 0, 12451935558776453120, 139667911854567, 0, 139667817901418}}, sa_flags = -1298457392, sa_restorer = 0x7f06b2cbcba4}
        sigs = {__val = {32, 0 <repeats 15 times>}}
#2  0x00007f06f82d240f in __assert_fail_base (fmt=0x7f06f2ac2d6a <error: Cannot access memory at address 0x7f06f2ac2d6a>, assertion=0x7f06b2cbcba4 "0", file=0x7f06b29b1cd0 "gen9_mfd.c", line=649, function=<optimized out>) at assert.c:92
        str = 0x7f06a009e1d0 "\220\335\a\240\006\177"
        total = 4096
#3  0x00007f06f82e1fb2 in __GI___assert_fail (assertion=assertion@entry=0x7f06b2cbcba4 "0", file=file@entry=0x7f06b29b1cd0 "gen9_mfd.c", line=line@entry=649, function=function@entry=0x7f06b29b23e0 <__PRETTY_FUNCTION__.14793> "gen9_hcpd_get_reference_picture_frame_id") at assert.c:101
No locals.
#4  0x00007f06b287af94 in gen9_hcpd_get_reference_picture_frame_id (frame_store=frame_store@entry=0x7f06a0079a70, ref_pic=<optimized out>, ref_pic=<optimized out>) at gen9_mfd.c:649
        i = <optimized out>
        __PRETTY_FUNCTION__ = "gen9_hcpd_get_reference_picture_frame_id"
#5  0x00007f06b287c94f in gen9_hcpd_get_collocated_ref_idx (ctx=0x7f06a80b8cf0, gen9_hcpd_context=0x7f06a0079a40, slice_param=0x7f06a0013a30, pic_param=0x7f06a00a08f0) at gen9_mfd.c:806
        ref_list = <optimized out>
        ref_pic = <optimized out>
        ref_list = <optimized out>
        ref_pic = <optimized out>
        __PRETTY_FUNCTION__ = "gen9_hcpd_get_collocated_ref_idx"
#6  gen9_hcpd_slice_state (ctx=0x7f06a80b8cf0, gen9_hcpd_context=0x7f06a0079a40, next_slice_param=0x7f06a00080e0, slice_param=0x7f06a0013a30, pic_param=0x7f06a00a08f0) at gen9_mfd.c:878
        slice_hor_pos = 0
        slice_ver_pos = 0
        collocated_ref_idx = <optimized out>
        collocated_from_l0_flag = <optimized out>
        sliceqp = 0
        batch = 0x7f06a000be80
        next_slice_hor_pos = 0
        next_slice_ver_pos = 8
        sliceqp_sign_flag = 0
        batch = <optimized out>
        slice_hor_pos = <optimized out>
        slice_ver_pos = <optimized out>
        next_slice_hor_pos = <optimized out>
        next_slice_ver_pos = <optimized out>
        collocated_ref_idx = <optimized out>
        collocated_from_l0_flag = <optimized out>
        sliceqp_sign_flag = <optimized out>
        sliceqp = <optimized out>
        __PRETTY_FUNCTION__ = "gen9_hcpd_slice_state"
#7  gen9_hcpd_hevc_decode_picture (ctx=0x7f06a80b8cf0, decode_state=0x7f06a80c7c68, gen9_hcpd_context=0x7f06a0079a40) at gen9_mfd.c:1027
        vaStatus = 0
        i965 = 0x7f06a80c5600
        batch = 0x7f06a000be80
        pic_param = 0x7f06a00a08f0
        slice_param = 0x7f06a0013a30
        next_slice_param = 0x7f06a00080e0
        next_slice_group_param = 0x7f06a00080e0
        slice_data_bo = <optimized out>
        i = 0
        j = 0
        __PRETTY_FUNCTION__ = "gen9_hcpd_hevc_decode_picture"
#8  0x00007f06d46222e0 in vaEndPicture (dpy=dpy@entry=0x7f06a80b8b80, context=context@entry=33554433) at va.c:1519
        va_status = 0
        ctx = 0x7f06a80b8cf0
#9  0x00007f06d468650e in gst_vaapi_picture_decode (picture=picture@entry=0x7f06a00a6550) at ../../../../gst-libs/gst/vaapi/gstvaapidecoder_objects.c:308
        iq_matrix = <optimized out>
        bitplane = <optimized out>
        huf_table = <optimized out>
        prob_table = <optimized out>
        va_display = 0x7f06a80b8b80
[... 360 lines cut ...]
        is_buffer = 1
        do_update_buffering = 0
        dropping = 0
        __PRETTY_FUNCTION__ = "gst_multi_queue_loop"
#41 0x00007f06f80b6321 in gst_task_func (task=0x560fb1623cb0) at gsttask.c:328
        lock = 0x7f06ac18cd70
        tself = 0x7f06a8014e30
        priv = 0x560fb1623c60
        __PRETTY_FUNCTION__ = "gst_task_func"
#42 0x00007f06f8f40353 in g_thread_pool_thread_proxy (data=<optimized out>) at ../../../glib/gthreadpool.c:308
        task = 0x560fb12757c0
        pool = 0x560fb0b416e0
#43 0x00007f06f8f3f98d in g_thread_proxy (data=0x7f06a8014e30) at ../../../glib/gthread.c:805
        thread = 0x7f06a8014e30
        __FUNCTION__ = "g_thread_proxy"
        _g_boolean_var_ = <optimized out>
#44 0x00007f06f7817182 in start_thread (arg=<optimized out>) at pthread_create.c:486
        ret = <optimized out>
        pd = <optimized out>
        now = <optimized out>
        unwind_buf = {cancel_jmp_buf = {{jmp_buf = {139666758035200, 7490950053542788013, 139666766420622, 139666766420623, 139666766420768, 139666758032448, -7352032257867220051, -7351884085033141331}, mask_was_saved = 0}}, priv = {pad = {0x0, 0x0, 0x0, 0x0}, data = {prev = 0x0, cleanup = 0x0, canceltype = 0}}}
        not_first_call = <optimized out>
#45 0x00007f06f83cafaf in clone () at ../sysdeps/unix/sysv/linux/x86_64/clone.S:95
No locals.
```

ejsc1, 2019-03-25:
I can always reproduce is error.

vanvugt, 2019-03-26:
Duplicate of #448.