intel/intel-vaapi-driver #452: VP9 Encoder causes GPU hang for tiled encode

sreerenjb, 2019-03-27:
How to reproduce the issue:

Instruct libva to encode with 2 columns tiles either in libva-utils or gstreamer-vaapi by setting the param:
pic_param->log2_tile_columns=1

Immediate GPU hang "i915 0000:00:02.0: Resetting vcs0 for hang on vcs0"

Brainiarc7, 2019-05-22:
Hello there,

With ffmpeg, can you reproduce the same when you pass the `-slices:v` option to the `vp9_vaapi` encoder?

sreerenjb, 2019-05-22:
I haven't tried. I don't have the ffmpeg vaapi stack. But It should be reproducible I guess because there is no specific logic involved with this, just the matter of assigning a non-zero value to tile_column attribute and it should hit the gpu hang. :)

xhaihao, 2019-06-18:
Yes, the devices supported by this driver don't support vp9 encoding with multiple tiles.

Brainiarc7, 2019-06-19:
Ouch. That hurts.