intel/intel-vaapi-driver #455: i965_drv_video.so has no function __vaDriverInit_0_32

tosonw, 2019-04-12:
On my ubuntu 16.04LTS desktop system, whenever the VAAPI driver is called, be it from vainfo or ffmpeg, I just get vaInitialize failed. 
Here's an example with output of what I run:
`$ vainfo
libva info: VA-API version 0.39.0
libva info: va_getDriverName() returns 0
libva info: User requested driver 'i965'
libva info: Trying to open /usr/lib/x86_64-linux-gnu/dri/i965_drv_video.so
libva error: /usr/lib/x86_64-linux-gnu/dri/i965_drv_video.so has no function __vaDriverInit_0_32
libva info: va_openDriver() returns -1
vaInitialize failed with error code -1 (unknown libva error),exit
`
Here are some commands and output showing that I have the necessary dependencies:
`# List installed libva packages
$ dpkg-query --showformat='${Package}: ${Version}\n' --show | grep libva
libva-dev: 1.7.0-1ubuntu0.1
libva-drm1: 1.7.0-1ubuntu0.1
libva-egl1: 1.7.0-1ubuntu0.1
libva-glx1: 1.7.0-1ubuntu0.1
libva-tpi1: 1.7.0-1ubuntu0.1
libva-wayland1: 1.7.0-1ubuntu0.1
libva-x11-1: 1.7.0-1ubuntu0.1
libva1: 1.7.0-1ubuntu0.1

# The vainfo installed
$ dpkg-query --showformat='${Package}: ${Version}\n' --show | grep vainfo
vainfo: 1.7.0-1ubuntu0.1

# List installed i915 Intel VAAPI driver
$ dpkg-query --showformat='${Package}: ${Version}\n' --show | grep i965
i965-va-driver: 1.7.0-1
`
Thanks!

tosonw, 2019-04-12:
> ```
> $ vainfo
> libva info: VA-API version 0.39.0
> libva info: va_getDriverName() returns 0
> libva info: User requested driver 'i965'
> libva info: Trying to open /usr/lib/x86_64-linux-gnu/dri/i965_drv_video.so
> libva info: Found init function __vaDriverInit_0_39
> vainfo: intel_driver.c:100: intel_driver_init: Assertion `(((struct drm_state *)(ctx)->drm_state)->auth_type == (VA_DRM_AUTH_DRI1)) || (((struct drm_state *)(ctx)->drm_state)->auth_type == (VA_DRM_AUTH_DRI2)) || (((struct drm_state *)(ctx)->drm_state)->auth_type == (VA_DRM_AUTH_CUSTOM))' failed.
> Aborted (core dumped)
> ```

tmm1, 2019-08-12:
It seems your driver was not compiled correctly.

You can try `nm /usr/lib/x86_64-linux-gnu/dri/i965_drv_video.so | grep vaDriverInit`

tosonw, 2019-08-16:
Thank you for your help.
Here are output showing:
`000000000006b120 T __vaDriverInit_1_5`
So what should the correct output look like?

xiaolin0199, 2020-01-09:
> Thank you for your help.
> Here are output showing:
> `000000000006b120 T __vaDriverInit_1_5`
> So what should the correct output look like?

Did you solve the problem ?