intel/intel-vaapi-driver #456: VP9 no longer plays on Chromium with Mesa 18.3.6, Kernel 5.0.7 and Coffee Lake

arcivanov, 2019-04-15:
This is going to be a really crappy bug report and I apologize for that in advance. 

A client has a remote system which is Intel NUC8i5BEK with Coffee Lake running Fedora 29 (5.0.7-200.fc29.x86_64) Mesa 18.3.6 with chromium-vaapi 73.0.3683.103-1.fc29 (https://admin.rpmfusion.org/pkgdb/package/free/chromium-vaapi/):

```
libva info: VA-API version 1.4.0
libva info: va_getDriverName() returns 0
libva info: Trying to open /usr/lib64/dri/i965_drv_video.so
libva info: Found init function __vaDriverInit_1_4
libva info: va_openDriver() returns 0
vainfo: VA-API version: 1.4 (libva 2.4.0)
vainfo: Driver version: Intel i965 driver for Intel(R) Coffee Lake - 2.3.0
vainfo: Supported profile and entrypoints
      VAProfileMPEG2Simple            : VAEntrypointVLD
      VAProfileMPEG2Simple            : VAEntrypointEncSlice
      VAProfileMPEG2Main              : VAEntrypointVLD
      VAProfileMPEG2Main              : VAEntrypointEncSlice
      VAProfileH264ConstrainedBaseline: VAEntrypointVLD
      VAProfileH264ConstrainedBaseline: VAEntrypointEncSlice
      VAProfileH264ConstrainedBaseline: VAEntrypointEncSliceLP
      VAProfileH264Main               : VAEntrypointVLD
      VAProfileH264Main               : VAEntrypointEncSlice
      VAProfileH264Main               : VAEntrypointEncSliceLP
      VAProfileH264High               : VAEntrypointVLD
      VAProfileH264High               : VAEntrypointEncSlice
      VAProfileH264High               : VAEntrypointEncSliceLP
      VAProfileH264MultiviewHigh      : VAEntrypointVLD
      VAProfileH264MultiviewHigh      : VAEntrypointEncSlice
      VAProfileH264StereoHigh         : VAEntrypointVLD
      VAProfileH264StereoHigh         : VAEntrypointEncSlice
      VAProfileVC1Simple              : VAEntrypointVLD
      VAProfileVC1Main                : VAEntrypointVLD
      VAProfileVC1Advanced            : VAEntrypointVLD
      VAProfileNone                   : VAEntrypointVideoProc
      VAProfileJPEGBaseline           : VAEntrypointVLD
      VAProfileJPEGBaseline           : VAEntrypointEncPicture
      VAProfileVP8Version0_3          : VAEntrypointVLD
      VAProfileVP8Version0_3          : VAEntrypointEncSlice
      VAProfileHEVCMain               : VAEntrypointVLD
      VAProfileHEVCMain               : VAEntrypointEncSlice
      VAProfileHEVCMain10             : VAEntrypointVLD
      VAProfileHEVCMain10             : VAEntrypointEncSlice
      VAProfileVP9Profile0            : VAEntrypointVLD
      VAProfileVP9Profile0            : VAEntrypointEncSlice
      VAProfileVP9Profile2            : VAEntrypointVLD
```

I'm locally running a system with an exact same version of software (kernel, Mesa, chromium-vaapi), except locally it's Dell Precision 7510 with Skylake GT2.

*****
Coffee Lake fails to play VP9 videos in Chromium & YT entirely.
Skylake plays VP9 videos in Chromium & YT no problem.
*****

The issue seems to have popped up somewhere between kernel 5.0.4 and 5.0.6 and continues in 5.0.7.

Again, I apologize for vagueness and lack of detail.

Plailect, 2019-04-17:
I am also experiencing this issue on Fedora 29 running kernel 5.0.7, Mesa 18.3.6 and the standard chromium package (73.0.3683.86) on my i7-8650U. I do not have any real information on when this bug started occurring (in terms of versions) as I just tried getting GPU accelerated decoding working for the first time today (by installing the `libva-intel-driver` and `libva-intel-hybrid-driver` packages). H264 decoding works just fine, but VP9 YouTube videos refuse to play now that the libva-intel-driver packages have been installed.

biswasab, 2019-05-01:
@arcivanov I think this bug is related to chromium. Since version 72, vp9 video has been broken.

arcivanov, 2019-05-02:
@biswasab Not on all hardware. 
I have Chromium VAAPI 73 and 74 on Skylake and VP9 works.
Same exact software setup on Coffee Lake NUC and VP9 doesn't work.

arcivanov, 2019-05-02:
Same issue with Fedora 30, Mesa 19.0.3

biswasab, 2019-05-03:
@arcivanov  Interesting. BTW, this bug is reported [downstream](https://bugzilla.redhat.com/show_bug.cgi?id=1694651).

arcivanov, 2019-05-03:
The fact that it works on some chipsets and not the others suggests a more complex issue than just "Chromium broke it". Could be there were valid changes in Chromium API calls that fail in Intel VAAPI.

x0wllaar, 2019-05-03:
Fedora 30, Chromium 73 on Coffee Lake - broken
Fedora 30, Chromium 73 on Cherry Trail - not broken.

Is [this issue](https://github.com/saiarcot895/chromium-ubuntu-build/issues/42) related to this?

arcivanov, 2019-05-03:
Commit list for Chromium VAAPI:
https://cs.chromium.org/chromium/src/media/gpu/vaapi/

freefreeno, 2019-05-13:
I am also having trouble playing only the Vp9 with the mojo decoder. I have Intel HD 620 with only mesa and intel-media-driver and the chromium-vaapi in Arch Linux. I can play everything but Vp9 with that mojo decoder. I have the variable set for the iHD driver and this is what i get when I type the vainfo

[josh@archkde ~]$ vainfo
vainfo: VA-API version: 1.4 (libva 2.4.1)
vainfo: Driver version: Intel iHD driver - 1.0.0
vainfo: Supported profile and entrypoints
      VAProfileNone                   : VAEntrypointVideoProc
      VAProfileNone                   : VAEntrypointStats
      VAProfileMPEG2Simple            : VAEntrypointVLD
      VAProfileMPEG2Simple            : VAEntrypointEncSlice
      VAProfileMPEG2Main              : VAEntrypointVLD
      VAProfileMPEG2Main              : VAEntrypointEncSlice
      VAProfileH264Main               : VAEntrypointVLD
      VAProfileH264Main               : VAEntrypointEncSlice
      VAProfileH264Main               : VAEntrypointFEI
      VAProfileH264Main               : VAEntrypointEncSliceLP
      VAProfileH264High               : VAEntrypointVLD
      VAProfileH264High               : VAEntrypointEncSlice
      VAProfileH264High               : VAEntrypointFEI
      VAProfileH264High               : VAEntrypointEncSliceLP
      VAProfileVC1Simple              : VAEntrypointVLD
      VAProfileVC1Main                : VAEntrypointVLD
      VAProfileVC1Advanced            : VAEntrypointVLD
      VAProfileJPEGBaseline           : VAEntrypointVLD
      VAProfileJPEGBaseline           : VAEntrypointEncPicture
      VAProfileH264ConstrainedBaseline: VAEntrypointVLD
      VAProfileH264ConstrainedBaseline: VAEntrypointEncSlice
      VAProfileH264ConstrainedBaseline: VAEntrypointFEI
      VAProfileH264ConstrainedBaseline: VAEntrypointEncSliceLP
      VAProfileVP8Version0_3          : VAEntrypointVLD
      VAProfileVP8Version0_3          : VAEntrypointEncSlice
      VAProfileHEVCMain               : VAEntrypointVLD
      VAProfileHEVCMain               : VAEntrypointEncSlice
      VAProfileHEVCMain               : VAEntrypointFEI
      VAProfileHEVCMain10             : VAEntrypointVLD
      VAProfileHEVCMain10             : VAEntrypointEncSlice
      VAProfileVP9Profile0            : VAEntrypointVLD
      VAProfileVP9Profile2            : VAEntrypointVLD

nphilipp, 2019-05-15:
Fedora 30, chromium-73.0.3683.86-2.fc30.x86_64:

* Gemini Lake (Zotac CI329 nano): broken
* Skylake (Lenovo ThinkPad P50): not broken

loonytoony0, 2019-05-21:
System: Debian Sid, kernel 5.2-rc1, every driver/firmware related to intel vaapi is installed from nonfree repo when possible
Chromium: 74.0.3729.108

Can confirm that it's still broken on Kaby Lake (Thinkpad T480s) even though vainfo shows that it's supported. Also vlc plays all vp9 profiles just fine.

```
libva info: VA-API version 1.4.0
libva info: va_getDriverName() returns 0
libva info: Trying to open /usr/lib/x86_64-linux-gnu/dri/i965_drv_video.so
libva info: Found init function __vaDriverInit_1_4
libva info: va_openDriver() returns 0
vainfo: VA-API version: 1.4 (libva 2.4.0)
vainfo: Driver version: Intel i965 driver for Intel(R) Kaby Lake - 2.3.0
vainfo: Supported profile and entrypoints
      VAProfileMPEG2Simple            : VAEntrypointVLD
      VAProfileMPEG2Simple            : VAEntrypointEncSlice
      VAProfileMPEG2Main              : VAEntrypointVLD
      VAProfileMPEG2Main              : VAEntrypointEncSlice
      VAProfileH264ConstrainedBaseline: VAEntrypointVLD
      VAProfileH264ConstrainedBaseline: VAEntrypointEncSlice
      VAProfileH264ConstrainedBaseline: VAEntrypointEncSliceLP
      VAProfileH264Main               : VAEntrypointVLD
      VAProfileH264Main               : VAEntrypointEncSlice
      VAProfileH264Main               : VAEntrypointEncSliceLP
      VAProfileH264High               : VAEntrypointVLD
      VAProfileH264High               : VAEntrypointEncSlice
      VAProfileH264High               : VAEntrypointEncSliceLP
      VAProfileH264MultiviewHigh      : VAEntrypointVLD
      VAProfileH264MultiviewHigh      : VAEntrypointEncSlice
      VAProfileH264StereoHigh         : VAEntrypointVLD
      VAProfileH264StereoHigh         : VAEntrypointEncSlice
      VAProfileVC1Simple              : VAEntrypointVLD
      VAProfileVC1Main                : VAEntrypointVLD
      VAProfileVC1Advanced            : VAEntrypointVLD
      VAProfileNone                   : VAEntrypointVideoProc
      VAProfileJPEGBaseline           : VAEntrypointVLD
      VAProfileJPEGBaseline           : VAEntrypointEncPicture
      VAProfileVP8Version0_3          : VAEntrypointVLD
      VAProfileVP8Version0_3          : VAEntrypointEncSlice
      VAProfileHEVCMain               : VAEntrypointVLD
      VAProfileHEVCMain               : VAEntrypointEncSlice
      VAProfileHEVCMain10             : VAEntrypointVLD
      VAProfileHEVCMain10             : VAEntrypointEncSlice
      VAProfileVP9Profile0            : VAEntrypointVLD
      VAProfileVP9Profile0            : VAEntrypointEncSlice
      VAProfileVP9Profile2            : VAEntrypointVLD
```

freefreeno, 2019-05-21:
Could someone here maybe help me with this. In Arch Linux on Kaby Lake Intel i5 7200u with Intel HD 620.
I did clean install just to make sure but let me try to explain. When set the variables in /etc/environment like this. ( I have huc and guc loaded ) ( Also early kms set like (i915) )

LIBVA_DRIVERS_PATH=/usr/lib/dri/iHD_drv_video.so
LIBVA_DRIVER_NAME=iHD

At this point I reboot and when I run the vainfo I get the error 1 as if there is NO variable set at all but if I remove the path variable and leave the LIBVA_DRIVER_NAME=iHD variable then and only then does the vainfo command give me my info. Also my videos play everything but Vp9 with hardware decoder. I have tried to find some reason that I can't set path but I have tried everything to no avail. When I run the the vainfo command as you see from my post above you do not see the 
Trying to open /usr/lib/dri/i965_drv_video.so
because I have to remove the path in /etc/environment

My videos do work with everything but Vp9 but I would like to understand what is going on. Also I have resorted to doing all this and then installing the h264ify extension to convert the Vp9 to to h264.
Is this an Arch Linux problem?? Please if anyone with more knowlege could help it would be greatly appreciated. Also I have tried to set the variables in the .bashrc insted and the exact same thing happens.

arcivanov, 2019-05-21:
`LIBVA_DRIVER_NAME=iHD`

You need to put it into /etc/profile (not sure if /etc/environment is the same on Arch) and `export` it for it to take.

arcivanov, 2019-05-21:
I found `LIBVA_DRIVERS_PATH` to be unnecessary. However, you're describing iHD, which is not intel-vaapi-driver but https://github.com/intel/media-driver

freefreeno, 2019-05-22:
It is the same on either for me. It works exactly the same for either driver for me.

freefreeno, 2019-05-22:
vainfo command will not work AT ALL IF YOU PUT PATH AND I THINK I EXPLAINED UP ABOVE THAT THAT I HAVE BEEN DOING EXACTLY WHAT YOU SAID.  etc/environment is where you put them in  Arch. 
BOTH DRIVERS. intel-vaapi or intel-media
of course I know they both have there own name. I explained very well above so just take iHD off my explanation and add i965.

uartie, 2019-05-22:
@sreerenjb can you make any sense of this?  Integration issue?

arcivanov, 2019-05-22:
Just curious, how would an integration issue affect one chipset but not the other?

uartie, 2019-05-22:
IDK.  I think one comment stated that VLC worked fine.  Perhaps if we had some other evidence without chromium that exhibit the same problem, it would help (e.g. does ffmpeg or gstreamer work?).  Nonetheless, @sreerenjb might have more knowledge about chromium + vaapi than I.

uartie, 2019-05-22:
I have not seen many issues with gstreamer or ffmpeg vp9 decode on 4.18 + CFL.  Thus, driver + kernel 5.0.x integration could be suspect too (I have not tried it).

biswasab, 2019-05-22:
>  Thus, driver + kernel 5.0.x integration could be suspect too (I have not tried it).

I can confirm that this issue does not depend on kernel version. The issue started appearing since chromium was updated from 72 to 73. Version 72 worked totally fine. I guess some new functions were introduced in chromium(like low power decoding/encoding).

freefreeno, 2019-05-22:
VLC does work fine. it is only Chromium giving me fits with now BOTH the intel-media-driver and the intel-vaapi-driver and I wish I knew why in Arch I can not enter the Path as an environment variable because then the Vainfo fails with error 1 and the my h264 hardware accel does not work at all. As it is I can decode h264 with the vaapi but I have use the hs26ify extension to convert Vp9 to h264 if I want to watch a youtube video. I am on Kernel 5.1.13-arch2-1-ARCH

x0wllaar, 2019-05-22:
VP9 in MPV works fine too.

loonytoony0, 2019-05-23:
> I have not seen many issues with gstreamer or ffmpeg vp9 decode on 4.18 + CFL. Thus, driver + kernel 5.0.x integration could be suspect too (I have not tried it).

It's absolutely no difference whether I try default debian sid 4.19 kernel or custom built 5.2rc1, I also tried 5.0 way before starting research on this issue and the behavior was identical, thus I don't think that kernel is involved.

loonytoony0, 2019-05-23:
Btw, sorry for misinformation, ThinkPad T480s is not kaby lake but coffee lake if that's still important.

Oxylibrium, 2019-06-02:
Can confirm that the issue is present - Arch Linux testing, `chromium-vaapi-bin` 74.0.3729.169-2, libva 1.4, Whiskey Lake i5-8265U on an ASUS Vivobook S430FA.

freefreeno, 2019-06-02:
@Oxylibrium Could you tell please tell me how you have your environment variables set and read all my post on this page and see if you have the same thing as me going on. I am in Arch also and have been trying the forum and everything but no one has even answered. I have a post over there called problems with hardware acceleration. Please if you would just give me some info of how and what you have done.

Oxylibrium, 2019-06-04:
@freefreeno Have both `libva-intel-driver` (this repo) and `intel-media-driver` (other one) installed. This driver is the default, as shown by vainfo. On Chromium, VP9 acceleration does not work with both drivers, and it just hangs indefinitely (tested on GNOME Wayland, GNOME Xorg, Plasma Xorg). However, H.264 decoding still works when I use h264ify. Also, VP9 decoding works with mpv (`--hwdec=vaapi`) and GNOME Videos (Totem media player, GStreamer). Unsure what is causing the issue.

freefreeno, 2019-06-04:
> @freefreeno Have both `libva-intel-driver` (this repo) and `intel-media-driver` (other one) installed. This driver is the default, as shown by vainfo. On Chromium, VP9 acceleration does not work [...]
Do you get the " Trying to open /usr/lib/dri/i965_drv_video.so" when running the vainfo command??????

Oxylibrium, 2019-06-04:
Output for vainfo verbatim:
```
[oxy@lithium ~]$ vainfo
vainfo: VA-API version: 1.4 (libva 2.4.1)
vainfo: Driver version: Intel i965 driver for Intel(R) Coffee Lake - 2.3.0
vainfo: Supported profile and entrypoints
      VAProfileMPEG2Simple            :	VAEntrypointVLD
      VAProfileMPEG2Simple            :	VAEntrypointEncSlice
      VAProfileMPEG2Main              :	VAEntrypointVLD
      VAProfileMPEG2Main              :	VAEntrypointEncSlice
      VAProfileH264ConstrainedBaseline:	VAEntrypointVLD
      VAProfileH264ConstrainedBaseline:	VAEntrypointEncSlice
      VAProfileH264ConstrainedBaseline:	VAEntrypointEncSliceLP
      VAProfileH264Main               :	VAEntrypointVLD
      VAProfileH264Main               :	VAEntrypointEncSlice
      VAProfileH264Main               :	VAEntrypointEncSliceLP
      VAProfileH264High               :	VAEntrypointVLD
      VAProfileH264High               :	VAEntrypointEncSlice
      VAProfileH264High               :	VAEntrypointEncSliceLP
      VAProfileH264MultiviewHigh      :	VAEntrypointVLD
      VAProfileH264MultiviewHigh      :	VAEntrypointEncSlice
      VAProfileH264StereoHigh         :	VAEntrypointVLD
      VAProfileH264StereoHigh         :	VAEntrypointEncSlice
      VAProfileVC1Simple              :	VAEntrypointVLD
      VAProfileVC1Main                :	VAEntrypointVLD
      VAProfileVC1Advanced            :	VAEntrypointVLD
      VAProfileNone                   :	VAEntrypointVideoProc
      VAProfileJPEGBaseline           :	VAEntrypointVLD
      VAProfileJPEGBaseline           :	VAEntrypointEncPicture
      VAProfileVP8Version0_3          :	VAEntrypointVLD
      VAProfileVP8Version0_3          :	VAEntrypointEncSlice
      VAProfileHEVCMain               :	VAEntrypointVLD
      VAProfileHEVCMain               :	VAEntrypointEncSlice
      VAProfileHEVCMain10             :	VAEntrypointVLD
      VAProfileHEVCMain10             :	VAEntrypointEncSlice
      VAProfileVP9Profile0            :	VAEntrypointVLD
      VAProfileVP9Profile0            :	VAEntrypointEncSlice
      VAProfileVP9Profile2            :	VAEntrypointVLD
```

freefreeno, 2019-06-04:
@Oxylibrium Ok I yours does not say the part about opening the driver either. I was just wondering. I guess Arch just don't say that part.

maximbaz, 2019-06-09:
FYI starting from Chromium 75, on some hardware (mine included) the video playback is broken for both VP9 and H264 with both i965 and iHD drivers, so something must have broken in the Chromium codebase itself. As soon as I revert to v74, H264 works again (but not VP9 as reported by this issue).

<details>
<summary>vainfo</summary>

```
❯ vainfo                   
vainfo: VA-API version: 1.4 (libva 2.4.1)
vainfo: Driver version: Intel i965 driver for Intel(R) Kaby Lake - 2.3.0
vainfo: Supported profile and entrypoints
      VAProfileMPEG2Simple            : VAEntrypointVLD
      VAProfileMPEG2Simple            : VAEntrypointEncSlice
      VAProfileMPEG2Main              : VAEntrypointVLD
      VAProfileMPEG2Main              : VAEntrypointEncSlice
      VAProfileH264ConstrainedBaseline: VAEntrypointVLD
      VAProfileH264ConstrainedBaseline: VAEntrypointEncSlice
      VAProfileH264ConstrainedBaseline: VAEntrypointEncSliceLP
      VAProfileH264Main               : VAEntrypointVLD
      VAProfileH264Main               : VAEntrypointEncSlice
      VAProfileH264Main               : VAEntrypointEncSliceLP
      VAProfileH264High               : VAEntrypointVLD
      VAProfileH264High               : VAEntrypointEncSlice
      VAProfileH264High               : VAEntrypointEncSliceLP
      VAProfileH264MultiviewHigh      : VAEntrypointVLD
      VAProfileH264MultiviewHigh      : VAEntrypointEncSlice
      VAProfileH264StereoHigh         : VAEntrypointVLD
      VAProfileH264StereoHigh         : VAEntrypointEncSlice
      VAProfileVC1Simple              : VAEntrypointVLD
      VAProfileVC1Main                : VAEntrypointVLD
      VAProfileVC1Advanced            : VAEntrypointVLD
      VAProfileNone                   : VAEntrypointVideoProc
      VAProfileJPEGBaseline           : VAEntrypointVLD
      VAProfileJPEGBaseline           : VAEntrypointEncPicture
      VAProfileVP8Version0_3          : VAEntrypointVLD
      VAProfileVP8Version0_3          : VAEntrypointEncSlice
      VAProfileHEVCMain               : VAEntrypointVLD
      VAProfileHEVCMain               : VAEntrypointEncSlice
      VAProfileHEVCMain10             : VAEntrypointVLD
      VAProfileHEVCMain10             : VAEntrypointEncSlice
      VAProfileVP9Profile0            : VAEntrypointVLD
      VAProfileVP9Profile0            : VAEntrypointEncSlice
      VAProfileVP9Profile2            : VAEntrypointVLD
                                                                    
❯ LIBVA_DRIVER_NAME=iHD vainfo  
vainfo: VA-API version: 1.4 (libva 2.4.1)
vainfo: Driver version: Intel iHD driver - 1.0.0
vainfo: Supported profile and entrypoints
      VAProfileNone                   : VAEntrypointVideoProc
      VAProfileNone                   : VAEntrypointStats
      VAProfileMPEG2Simple            : VAEntrypointVLD
      VAProfileMPEG2Simple            : VAEntrypointEncSlice
      VAProfileMPEG2Main              : VAEntrypointVLD
      VAProfileMPEG2Main              : VAEntrypointEncSlice
      VAProfileH264Main               : VAEntrypointVLD
      VAProfileH264Main               : VAEntrypointEncSlice
      VAProfileH264Main               : VAEntrypointFEI
      VAProfileH264Main               : VAEntrypointEncSliceLP
      VAProfileH264High               : VAEntrypointVLD
      VAProfileH264High               : VAEntrypointEncSlice
      VAProfileH264High               : VAEntrypointFEI
      VAProfileH264High               : VAEntrypointEncSliceLP
      VAProfileVC1Simple              : VAEntrypointVLD
      VAProfileVC1Main                : VAEntrypointVLD
      VAProfileVC1Advanced            : VAEntrypointVLD
      VAProfileJPEGBaseline           : VAEntrypointVLD
      VAProfileJPEGBaseline           : VAEntrypointEncPicture
      VAProfileH264ConstrainedBaseline: VAEntrypointVLD
      VAProfileH264ConstrainedBaseline: VAEntrypointEncSlice
      VAProfileH264ConstrainedBaseline: VAEntrypointFEI
      VAProfileH264ConstrainedBaseline: VAEntrypointEncSliceLP
      VAProfileVP8Version0_3          : VAEntrypointVLD
      VAProfileVP8Version0_3          : VAEntrypointEncSlice
      VAProfileHEVCMain               : VAEntrypointVLD
      VAProfileHEVCMain               : VAEntrypointEncSlice
      VAProfileHEVCMain               : VAEntrypointFEI
      VAProfileHEVCMain10             : VAEntrypointVLD
      VAProfileHEVCMain10             : VAEntrypointEncSlice
      VAProfileVP9Profile0            : VAEntrypointVLD
      VAProfileVP9Profile2            : VAEntrypointVLD
```

</details>

loonytoony0, 2019-06-13:
Agree with previous comment. After updating to chromium 75 it's now unable to play damn H264.

biswasab, 2019-06-15:
Just updated to 75, h.264 acceleration works fine on AMDGPU driver. So it's a Intel driver problem.

arcivanov, 2019-06-16:
FYI, there are tons of changes in the Chromium VAAPI in 77.x release train, including new management factories etc.

https://github.com/chromium/chromium/commit/c6d13e9dcb2fc510c0f7de1532d351e1c4ee6a12#diff-bccebd678dde7a9e7698c5e4b166d1d7

biswasab, 2019-06-21:
@arcivanov Yes I know. I am following the development.

BTW, Has someone tested chromium vaapi on the experimental  Intel "Iris" Gallium3D driver on Mesa 19.1?

arcivanov, 2019-06-21:
Chromium is changing flags from "linux" to "chromeos" to pretend that Chromeos is not Linux. The comment also intimates that there may be changes that make the Chromeos VAAPI implementation incompatible with Linux at some point in the future.

https://github.com/chromium/chromium/commit/692de4cb791a78f55ed626ebed48ace7ed9de66e#diff-538769d4eb59e43e55b2f616b182967c

arcivanov, 2019-06-21:
@biswasab From your experience, how much work do you think would it be to try to stick chromium/media/gpu from HEAD into the stable train (75 at present)?

biswasab, 2019-06-22:
@arcivanov Reverting those changes will be a piece of cake for me. I'm not worried about that. I worried about what else they will break in the process.

BTW, how much they shout about it they cannot change the fact that chromeos is still linux, whether they use a outdated libva or a up to date stable one.

Lessaj, 2019-06-23:
> Just updated to 75, h.264 acceleration works fine on AMDGPU driver. So it's an Intel driver problem.

Just updated to 75 as well, chromium-vaapi-bin on arch linux, and h.264 is NOT working for me with AMDGPU driver, can't load anything at all, however VP9 works on YouTube. Having said that I also have intel hardware in my system but I am not currently using it for display output, i915 is still required to be loaded for audio hardware, this might be causing the issue. I am using mesagit package group from lcarlier which is currently Mesa 19.2.

`vainfo: VA-API version: 1.4 (libva 2.4.1)
vainfo: Driver version: Mesa Gallium driver 19.2.0-devel for Radeon RX Vega (VEGA10, DRM 3.30.0, 5.1.14-1-ck, LLVM 9.0.0)
vainfo: Supported profile and entrypoints
      VAProfileMPEG2Simple            :	VAEntrypointVLD
      VAProfileMPEG2Main              :	VAEntrypointVLD
      VAProfileVC1Simple              :	VAEntrypointVLD
      VAProfileVC1Main                :	VAEntrypointVLD
      VAProfileVC1Advanced            :	VAEntrypointVLD
      VAProfileH264ConstrainedBaseline:	VAEntrypointVLD
      VAProfileH264ConstrainedBaseline:	VAEntrypointEncSlice
      VAProfileH264Main               :	VAEntrypointVLD
      VAProfileH264Main               :	VAEntrypointEncSlice
      VAProfileH264High               :	VAEntrypointVLD
      VAProfileH264High               :	VAEntrypointEncSlice
      VAProfileHEVCMain               :	VAEntrypointVLD
      VAProfileHEVCMain               :	VAEntrypointEncSlice
      VAProfileHEVCMain10             :	VAEntrypointVLD
      VAProfileNone                   :	VAEntrypointVideoProc
`

Realex-fire, 2019-07-01:
I confirm this problem on Intel Kaby Lake. Chromium 75 does not hardware decode H264 and VP9 video.
My system: openSUSE Tumbleweed, Linux 5.1.10, Mesa 19.1
`
libva info: VA-API version 1.4.0
libva info: va_getDriverName() returns 0
libva info: Trying to open /usr/lib64/dri/i965_drv_video.so
libva info: Found init function __vaDriverInit_1_4
libva info: va_openDriver() returns 0
vainfo: VA-API version: 1.4 (libva 2.3.0)
vainfo: Driver version: Intel i965 driver for Intel(R) Kaby Lake - 2.3.0
vainfo: Supported profile and entrypoints
      VAProfileMPEG2Simple            : VAEntrypointVLD
      VAProfileMPEG2Simple            : VAEntrypointEncSlice
      VAProfileMPEG2Main              : VAEntrypointVLD
      VAProfileMPEG2Main              : VAEntrypointEncSlice
      VAProfileH264ConstrainedBaseline: VAEntrypointVLD
      VAProfileH264ConstrainedBaseline: VAEntrypointEncSlice
      VAProfileH264ConstrainedBaseline: VAEntrypointEncSliceLP
      VAProfileH264Main               : VAEntrypointVLD
      VAProfileH264Main               : VAEntrypointEncSlice
      VAProfileH264Main               : VAEntrypointEncSliceLP
      VAProfileH264High               : VAEntrypointVLD
      VAProfileH264High               : VAEntrypointEncSlice
      VAProfileH264High               : VAEntrypointEncSliceLP
      VAProfileH264MultiviewHigh      : VAEntrypointVLD
      VAProfileH264MultiviewHigh      : VAEntrypointEncSlice
      VAProfileH264StereoHigh         : VAEntrypointVLD
      VAProfileH264StereoHigh         : VAEntrypointEncSlice
      VAProfileVC1Simple              : VAEntrypointVLD
      VAProfileVC1Main                : VAEntrypointVLD
      VAProfileVC1Advanced            : VAEntrypointVLD
      VAProfileNone                   : VAEntrypointVideoProc
      VAProfileJPEGBaseline           : VAEntrypointVLD
      VAProfileJPEGBaseline           : VAEntrypointEncPicture
      VAProfileVP8Version0_3          : VAEntrypointVLD
      VAProfileVP8Version0_3          : VAEntrypointEncSlice
      VAProfileHEVCMain               : VAEntrypointVLD
      VAProfileHEVCMain               : VAEntrypointEncSlice
      VAProfileHEVCMain10             : VAEntrypointVLD
      VAProfileHEVCMain10             : VAEntrypointEncSlice
      VAProfileVP9Profile0            : VAEntrypointVLD
      VAProfileVP9Profile0            : VAEntrypointEncSlice
      VAProfileVP9Profile2            : VAEntrypointVLD
`
Templary solution: disable #disable-accelerated-video-decode in chrome://flags to enable software video decoding.

Realex-fire, 2019-07-01:
I confirm this problem on Intel Kaby Lake. Chromium 75 does not hardware decode H264 and VP9 video.
My system: openSUSE Tumbleweed, Linux 5.1.10, Mesa 19.1

```
libva info: VA-API version 1.4.0
libva info: va_getDriverName() returns 0
libva info: Trying to open /usr/lib64/dri/i965_drv_video.so
libva info: Found init function __vaDriverInit_1_4
libva info: va_openDriver() returns 0
vainfo: VA-API version: 1.4 (libva 2.3.0)
vainfo: Driver version: Intel i965 driver for Intel(R) Kaby Lake - 2.3.0
vainfo: Supported profile and entrypoints
      VAProfileMPEG2Simple            : VAEntrypointVLD
      VAProfileMPEG2Simple            : VAEntrypointEncSlice
      VAProfileMPEG2Main              : VAEntrypointVLD
      VAProfileMPEG2Main              : VAEntrypointEncSlice
      VAProfileH264ConstrainedBaseline: VAEntrypointVLD
      VAProfileH264ConstrainedBaseline: VAEntrypointEncSlice
      VAProfileH264ConstrainedBaseline: VAEntrypointEncSliceLP
      VAProfileH264Main               : VAEntrypointVLD
      VAProfileH264Main               : VAEntrypointEncSlice
      VAProfileH264Main               : VAEntrypointEncSliceLP
      VAProfileH264High               : VAEntrypointVLD
      VAProfileH264High               : VAEntrypointEncSlice
      VAProfileH264High               : VAEntrypointEncSliceLP
      VAProfileH264MultiviewHigh      : VAEntrypointVLD
      VAProfileH264MultiviewHigh      : VAEntrypointEncSlice
      VAProfileH264StereoHigh         : VAEntrypointVLD
      VAProfileH264StereoHigh         : VAEntrypointEncSlice
      VAProfileVC1Simple              : VAEntrypointVLD
      VAProfileVC1Main                : VAEntrypointVLD
      VAProfileVC1Advanced            : VAEntrypointVLD
      VAProfileNone                   : VAEntrypointVideoProc
      VAProfileJPEGBaseline           : VAEntrypointVLD
      VAProfileJPEGBaseline           : VAEntrypointEncPicture
      VAProfileVP8Version0_3          : VAEntrypointVLD
      VAProfileVP8Version0_3          : VAEntrypointEncSlice
      VAProfileHEVCMain               : VAEntrypointVLD
      VAProfileHEVCMain               : VAEntrypointEncSlice
      VAProfileHEVCMain10             : VAEntrypointVLD
      VAProfileHEVCMain10             : VAEntrypointEncSlice
      VAProfileVP9Profile0            : VAEntrypointVLD
      VAProfileVP9Profile0            : VAEntrypointEncSlice
      VAProfileVP9Profile2            : VAEntrypointVLD
```
Templary solution: disable #disable-accelerated-video-decode in chrome://flags to enable software video decoding.

maximbaz, 2019-07-03:
@akarshanbiswas shared a [patch](https://aur.archlinux.org/cgit/aur.git/tree/vaapi-fix.patch?h=chromium-vaapi) that fixed both H264 and VP9 for me in the latest stable Chromium!

akarshanbiswas, 2019-07-03:
I guess it's all because of this: https://bugs.chromium.org/p/chromium/issues/detail?id=912295

VPP disabled for Nvidia.

ronasimi, 2019-07-03:
Using the new build 75.0.3770.100 and confirming acceleration working on Sandybridge mobile.

Lessaj, 2019-07-04:
75.0.3770.100 now working for me as well.

akarshanbiswas, 2019-07-04:
@arcivanov  You can test it from [here](http://koji.rpmfusion.org/koji/buildinfo?buildID=11624) on Fedora.

arcivanov, 2019-07-04:
@akarshanbiswas chromium-vaapi-75.0.3770.100-1.fc30.x86_64 **not working** on Coffee Lake (Iris Plus 655) - neither MP4 nor VP9 work on YT.

arcivanov, 2019-07-04:
@akarshanbiswas I confirm chromium-vaapi-75.0.3770.100-2.fc30.x86_64 **works** on Coffee Lake (Iris Plus 655)

jstkdng, 2019-07-18:
Just adding my piece of information.
VP9 is not working with chromium 75.0.3770.142 under kaby lake.
Playback works if I disable video acceleration though.

It works with vdpau with my nvidia gpu though. Haven't tested amdgpu.

akarshanbiswas, 2019-07-18:
@jstkdng Which distro?

jstkdng, 2019-07-18:
I'm on archlinux. Managed to fix the problem with this [patch](https://git.vin.ovh/jk/ungoogled-chromium-archlinux/raw/branch/master/vaapi-fix.patch).
And aparently it was written by you @akarshanbiswas m8, thanks!
Oh, btw, that patch also works with intel media driver.

tbergeron, 2019-11-20:
I believe I've been having similar issues on Ubuntu 19.04 on kernel 5.3.7.
To have proper gpu support I have to use `--ignore-gpu-blacklist` in Chromium. It's way snappier with that flag but for some reason webm/vp9 videos do not play at all. Without that flag, they do play.

I noticed that without `--ignore-gpu-blacklist` WebM videos are decoded with FFMPEG (when looking in `chrome://media-internals`) and are playing. But when using the flag, `MojoVideoDecoder` is handling the video, I see the length but the video never plays.

At first I thought it was my video drivers, I use `ppa:oibaf/graphics-drivers` I tried without these and same happens. I really don't know if it's the driver or chromium but that's the only discussion I found yet about the issue.

Here's my `vainfo`:

```
vainfo: VA-API version: 1.4 (libva 2.4.0)
vainfo: Driver version: Intel i965 driver for Intel(R) Kaby Lake - 2.3.0
vainfo: Supported profile and entrypoints
      VAProfileMPEG2Simple            :	VAEntrypointVLD
      VAProfileMPEG2Simple            :	VAEntrypointEncSlice
      VAProfileMPEG2Main              :	VAEntrypointVLD
      VAProfileMPEG2Main              :	VAEntrypointEncSlice
      VAProfileH264ConstrainedBaseline:	VAEntrypointVLD
      VAProfileH264ConstrainedBaseline:	VAEntrypointEncSlice
      VAProfileH264ConstrainedBaseline:	VAEntrypointEncSliceLP
      VAProfileH264Main               :	VAEntrypointVLD
      VAProfileH264Main               :	VAEntrypointEncSlice
      VAProfileH264Main               :	VAEntrypointEncSliceLP
      VAProfileH264High               :	VAEntrypointVLD
      VAProfileH264High               :	VAEntrypointEncSlice
      VAProfileH264High               :	VAEntrypointEncSliceLP
      VAProfileH264MultiviewHigh      :	VAEntrypointVLD
      VAProfileH264MultiviewHigh      :	VAEntrypointEncSlice
      VAProfileH264StereoHigh         :	VAEntrypointVLD
      VAProfileH264StereoHigh         :	VAEntrypointEncSlice
      VAProfileVC1Simple              :	VAEntrypointVLD
      VAProfileVC1Main                :	VAEntrypointVLD
      VAProfileVC1Advanced            :	VAEntrypointVLD
      VAProfileNone                   :	VAEntrypointVideoProc
      VAProfileJPEGBaseline           :	VAEntrypointVLD
      VAProfileJPEGBaseline           :	VAEntrypointEncPicture
      VAProfileVP8Version0_3          :	VAEntrypointVLD
      VAProfileVP8Version0_3          :	VAEntrypointEncSlice
      VAProfileHEVCMain               :	VAEntrypointVLD
      VAProfileHEVCMain               :	VAEntrypointEncSlice
      VAProfileHEVCMain10             :	VAEntrypointVLD
      VAProfileHEVCMain10             :	VAEntrypointEncSlice
      VAProfileVP9Profile0            :	VAEntrypointVLD
      VAProfileVP9Profile0            :	VAEntrypointEncSlice
      VAProfileVP9Profile2            :	VAEntrypointVLD
```

Feel free to ask for more debug info, thanks

freefreeno, 2019-11-20:
Ok just to let everyone know what I have done and what is working in Arch Linux. I have only the modesetting  driver as my default on i7 9750H Coffee Lake with Intel HD 630. I have the intel-media-driver installed and for those of you on Ubuntu you need to install the intel-media-driver non free. Also you need to enable huc and guc for full functions for the intel-media-driver. I can play any format this way. Ok for my question that I hope someone can answer. The Intel Iris driver is about to be released. Is this the intel-vaapi-driver???? All I know is it says it is the i965 driver. I would like to use the iris driver for my Intel graphics if this possible but I don't even know if this is the same driver. If it is the same driver then how do we use it. It is supposed to way better so if anyone can feel me in on Intel Iris info then this would be appreciated.