intel/intel-vaapi-driver #462: i915/SKL GPU HANG: ecode 9:0:0x00000000/ecode 9:2:0xa8dfbffd, in chromium-vaapi, reason: hang on vcs0, action: reset hang on vcs0

arcivanov, 2019-05-20:
I'm moving this here at the request of the kernel i915 team. There are ample i915 error dumps attached to the below bug from 5.0.x and 5.1.x kernels with and without drm-tip.

Skylake GPU hangs cause complete freezing of the entire kernel with no recovery possible.

https://bugs.freedesktop.org/show_bug.cgi?id=109949

arcivanov, 2019-05-24:
@uartie could you please triage this?

uartie, 2019-05-24:
What codec/input stream are you experiencing this with?  Can you share the video input?

arcivanov, 2019-05-24:
It's Chromium, YT/Hulu/Netflix, so different codecs, so MP4/VP9/whatever Hulu/Netflix uses. I can't speak for other reporters but I assume the same. There are GPU error dumps a-plenty attached.

arcivanov, 2019-05-24:
I personally don't believe the full kernel lockup can be legitimately caused by userland (it's a terrible kernel module design if true and a huge security concern), but i915 kernel team seems to be dead set on bumping this over to you, guys.

uartie, 2019-05-24:
Maybe I missed it on the linked bug, but could you provide some additional software versions (e.g. vaapi,  vaapi driver, chromium, etc.)?

uartie, 2019-05-24:
@sreerenjb since chromium is the only known reproducing starting point, could you point to someone in chromium-land that can start a deeper root cause?

arcivanov, 2019-05-24:
I'll be happy to. It happened pretty much with all Chromium-vaapi (https://admin.rpmfusion.org/pkgdb/package/free/chromium-vaapi/) 7x versions.

```
$ vainfo
libva info: VA-API version 1.4.1
libva info: va_getDriverName() returns 0
libva info: Trying to open /usr/lib64/dri/i965_drv_video.so
libva info: Found init function __vaDriverInit_1_4
libva info: va_openDriver() returns 0
vainfo: VA-API version: 1.4 (libva 2.4.0)
vainfo: Driver version: Intel i965 driver for Intel(R) Skylake - 2.3.0
vainfo: Supported profile and entrypoints
      VAProfileMPEG2Simple            : VAEntrypointVLD
      VAProfileMPEG2Simple            : VAEntrypointEncSlice
      VAProfileMPEG2Main              : VAEntrypointVLD
      VAProfileMPEG2Main              : VAEntrypointEncSlice
      VAProfileH264ConstrainedBaseline: VAEntrypointVLD
      VAProfileH264ConstrainedBaseline: VAEntrypointEncSlice
      VAProfileH264ConstrainedBaseline: VAEntrypointEncSliceLP
      VAProfileH264ConstrainedBaseline: VAEntrypointFEI
      VAProfileH264ConstrainedBaseline: VAEntrypointStats
      VAProfileH264Main               : VAEntrypointVLD
      VAProfileH264Main               : VAEntrypointEncSlice
      VAProfileH264Main               : VAEntrypointEncSliceLP
      VAProfileH264Main               : VAEntrypointFEI
      VAProfileH264Main               : VAEntrypointStats
      VAProfileH264High               : VAEntrypointVLD
      VAProfileH264High               : VAEntrypointEncSlice
      VAProfileH264High               : VAEntrypointEncSliceLP
      VAProfileH264High               : VAEntrypointFEI
      VAProfileH264High               : VAEntrypointStats
      VAProfileH264MultiviewHigh      : VAEntrypointVLD
      VAProfileH264MultiviewHigh      : VAEntrypointEncSlice
      VAProfileH264StereoHigh         : VAEntrypointVLD
      VAProfileH264StereoHigh         : VAEntrypointEncSlice
      VAProfileVC1Simple              : VAEntrypointVLD
      VAProfileVC1Main                : VAEntrypointVLD
      VAProfileVC1Advanced            : VAEntrypointVLD
      VAProfileNone                   : VAEntrypointVideoProc
      VAProfileJPEGBaseline           : VAEntrypointVLD
      VAProfileJPEGBaseline           : VAEntrypointEncPicture
      VAProfileVP8Version0_3          : VAEntrypointVLD
      VAProfileVP8Version0_3          : VAEntrypointEncSlice
      VAProfileHEVCMain               : VAEntrypointVLD
      VAProfileHEVCMain               : VAEntrypointEncSlice
      VAProfileVP9Profile0            : VAEntrypointVLD
```
```
May 22 13:03:30 hostname kernel: Command line: BOOT_IMAGE=(hd0,gpt2)/vmlinuz-5.0.16-300.fc30.x86_64 root=/dev/mapper/fedora-root ro resume=/dev/mapper/fedora-swap rd.lvm.lv=fedora/root rd.lvm.lv=fedora/swap rootflags=discard rhgb i915.enable_dc=2 i915.disable_power_well=0 i915.enable_fbc=1 i915.enable_guc=3 i915.enable_dpcd_backlight=1 l1tf=flush rd.driver.blacklist=nouveau
May 22 13:03:30 hostname kernel: Kernel command line: BOOT_IMAGE=(hd0,gpt2)/vmlinuz-5.0.16-300.fc30.x86_64 root=/dev/mapper/fedora-root ro resume=/dev/mapper/fedora-swap rd.lvm.lv=fedora/root rd.lvm.lv=fedora/swap rootflags=discard rhgb i915.enable_dc=2 i915.disable_power_well=0 i915.enable_fbc=1 i915.enable_guc=3 i915.enable_dpcd_backlight=1 l1tf=flush rd.driver.blacklist=nouveau
May 22 13:03:31 hostname kernel: fb0: switching to inteldrmfb from EFI VGA
May 22 13:03:31 hostname kernel: [drm] Replacing VGA console driver
May 22 13:03:31 hostname kernel: [drm] Supports vblank timestamp caching Rev 2 (21.10.2013).
May 22 13:03:31 hostname kernel: [drm] Driver supports precise vblank timestamp query.
May 22 13:03:31 hostname kernel: i915 0000:00:02.0: vgaarb: changed VGA decodes: olddecodes=io+mem,decodes=none:owns=io+mem
May 22 13:03:31 hostname kernel: [drm] Finished loading DMC firmware i915/skl_dmc_ver1_27.bin (v1.27)
May 22 13:03:31 hostname kernel: [drm] HuC: Loaded firmware i915/skl_huc_ver01_07_1398.bin (version 1.7)
May 22 13:03:31 hostname kernel: [drm] GuC: Loaded firmware i915/skl_guc_ver9_33.bin (version 9.33)
May 22 13:03:31 hostname kernel: i915 0000:00:02.0: GuC firmware version 9.33
May 22 13:03:31 hostname kernel: i915 0000:00:02.0: GuC submission enabled
May 22 13:03:31 hostname kernel: i915 0000:00:02.0: HuC enabled
May 22 13:03:31 hostname kernel: [drm] Initialized i915 1.6.0 20181204 for 0000:00:02.0 on minor 0
May 22 13:03:31 hostname kernel: fbcon: inteldrmfb (fb0) is primary device
May 22 13:03:31 hostname kernel: i915 0000:00:02.0: fb0: inteldrmfb frame buffer device
May 22 13:03:32 hostname kernel: RAPL PMU: hw unit of domain pp1-gpu 2^-14 Joules
May 22 13:03:32 hostname kernel: snd_hda_intel 0000:00:1f.3: bound 0000:00:02.0 (ops i915_audio_component_bind_ops [i915])
May 22 13:59:35 hostname kernel: [drm] Reducing the compressed framebuffer size. This may lead to less power savings than a non-reduced-size. Try to increase stolen memory size if available in BIOS.
May 22 14:16:50 hostname kernel: [drm] Reducing the compressed framebuffer size. This may lead to less power savings than a non-reduced-size. Try to increase stolen memory size if available in BIOS.
May 22 14:16:54 hostname kernel: [drm] Reducing the compressed framebuffer size. This may lead to less power savings than a non-reduced-size. Try to increase stolen memory size if available in BIOS.
[... cut ...]
May 23 18:23:57 hostname kernel: [drm] Reducing the compressed framebuffer size. This may lead to less power savings than a non-reduced-size. Try to increase stolen memory size if available in BIOS.
May 23 19:34:16 hostname kernel: [drm] Reducing the compressed framebuffer size. This may lead to less power savings than a non-reduced-size. Try to increase stolen memory size if available in BIOS.
May 23 22:22:23 hostname kernel: [drm] Reducing the compressed framebuffer size. This may lead to less power savings than a non-reduced-size. Try to increase stolen memory size if available in BIOS.
```

arcivanov, 2019-05-24:
Importantly, I'm running exact same configuration of software (Fedora 30, Chromium-vaapi, chromium settings and flags) on CoffeeLake Intel NUC for a client, and there are no lockup issues whatsoever. I would've thought it's my hardware, but other people are reporting the lockups with videos running in exactly the same fashion.

arcivanov, 2019-05-24:
Another note, the other reporters also are experiencing this issue on Skylake with newer DRM driver:

```
Mär 10 02:47:54 host kernel: [drm] Finished loading DMC firmware i915/skl_dmc_ver1_27.bin (v1.27)
Mär 10 02:47:54 host kernel: [drm] Initialized i915 1.6.0 20190220 for 0000:00:02.0 on minor 0
```

arcivanov, 2019-05-24:
Since you asked for a video this video caused 2 hard lockups within an hour: https://www.youtube.com/watch?v=KcTnNJtGnNE

[image]

arcivanov, 2019-05-24:
[image]

cynecx, 2019-06-08:
Has there been any progress in identifying the cause of this? I am experiencing the exact same problem except that I am unable to recover anything because of kernel is basically dead.

arcivanov, 2019-06-10:
@cynecx I've been pushing for kernel driver to recognize this as a kernel problem as well: there are no legitimate reasons why userspace should ever be able to kill a kernel, regardless of how much userspace misbehaves.

https://bugs.freedesktop.org/show_bug.cgi?id=109949#c34

arcivanov, 2019-06-16:
These issues may be fixed in Chromium in 76 and 77 trains, but it's hard to say whether GPU hangs are specifically related to the below issue.

https://github.com/chromium/chromium/commit/a2f0b776bede765f5901276c5698986bc13fa1c8#diff-bccebd678dde7a9e7698c5e4b166d1d7

akarshanbiswas, 2019-07-27:
@arcivanov I will backport it.

agaskins, 2019-12-21:
I know this is old, but after dealing with these same issues for months I switched to intel-media-driver and these constant errors went away for me, as well as a problem I had with _seemingly_ random artifacts and and system hangs. I'm not sure it's relevant to this, but I am on a very similar system and using Vivaldi (which is also chrome-based) so it seems plausible. I'm not savvy enough to know why it fixed this - just that it did. If there's anything I could do to shed some light on it I'd be happy to run some commands and post back... or whatever. There's a lot of folks with similar issues if you dig around the interwebs, and I wonder if this is all related or just a common error that encompasses multiple issues.