intel/intel-vaapi-driver #467: Chromium: libva error: /usr/lib/dri/i965_drv_video.so init failed

braderhart, 2019-05-23:
OS: Arch with kernel 5.1.4
Browser: [Chromium 74.0.3729.15 w/VAAPI](https://aur.archlinux.org/packages/chromium-vaapi-bin)

Command: `LIBGL_DRI3_DISABLE=1 LIBVA_DRIVER_NAME=i965 chromium`:

```
libva error: /usr/lib/dri/i965_drv_video.so init failed
[17617:17617:0522/194033.622300:ERROR:vaapi_wrapper.cc(335)] vaInitialize failed: unknown libva error
[17617:17617:0522/194033.795075:ERROR:sandbox_linux.cc(368)] InitializeSandbox() called with multiple threads in process gpu-process.
[17617:17617:0522/194034.032152:ERROR:buffer_manager.cc(488)] [.DisplayCompositor]GL ERROR :GL_INVALID_OPERATION : glBufferData: <- error from previous GL command
```

braderhart, 2019-05-23:
Same error without `LIBGL_DRI3_DISABLE=1`:

```
LIBVA_DRIVER_NAME=i965 chromium
libva error: /usr/lib/dri/i965_drv_video.so init failed
[26945:26945:0522/194656.904548:ERROR:vaapi_wrapper.cc(335)] vaInitialize failed: unknown libva error
[26945:26945:0522/194656.931465:ERROR:sandbox_linux.cc(368)] InitializeSandbox() called with multiple threads in process gpu-process.
[26945:26945:0522/194657.325322:ERROR:buffer_manager.cc(488)] [.DisplayCompositor]GL ERROR :GL_INVALID_OPERATION : glBufferData: <- error from previous GL command
[26945:26945:0522/194701.457471:ERROR:gl_surface_presentation_helper.cc(244)] GetVSyncParametersIfAvailable() failed for 1 times!
[26945:26945:0522/194701.462119:ERROR:gl_surface_presentation_helper.cc(244)] GetVSyncParametersIfAvailable() failed for 2 times!
```

xhaihao, 2019-05-23:
does vainfo work for your?

braderhart, 2019-05-23:
Yes, that works fine.

L3P3, 2020-07-07:
When I enter vainfo on my machine, I get:
```error: XDG_RUNTIME_DIR not set in the environment.
libva info: VA-API version 1.8.0
libva info: Trying to open /usr/lib/x86_64-linux-gnu/dri/iHD_drv_video.so
libva info: Found init function __vaDriverInit_1_7
libva error: /usr/lib/x86_64-linux-gnu/dri/iHD_drv_video.so init failed
libva info: va_openDriver() returns 1
libva info: Trying to open /usr/lib/x86_64-linux-gnu/dri/i965_drv_video.so
libva info: Found init function __vaDriverInit_1_5
libva error: /usr/lib/x86_64-linux-gnu/dri/i965_drv_video.so init failed
libva info: va_openDriver() returns -1
vaInitialize failed with error code -1 (unknown libva error),exit```

pavel-the-best, 2020-08-27:
Same for me, version 2.4.1

closed 2022-01-29 by aijai9hu

Diegovsky, 2022-02-10:
I see. Is this fixed tho?

Diegovsky, 2022-02-11:
Cool, can you describe what you did? I'm habing this happen in a hybrid setup of Intel graphics x nvidia so I think it's the intel driver because nvidia does not implement VA.

Atemu, 2022-02-11:
No idea, sorry. It was a long time ago and I neither have an Intel nor Nvidia GPU anymore.

Diegovsky, 2022-02-11:
That's ok, thanks :)