intel/intel-vaapi-driver #469: Support Ice Lake architecture

EwoutH, 2019-05-28:
Intel [formally announced](https://newsroom.intel.com/news-releases/2019-computex-intel-disclosures/) their 10th Gen processors today, based on the Ice Lake (ICL) micro-architecture. It would be nice if support for their new media encoder could be added.

The [Intel Media Driver](https://github.com/intel/media-driver) already supports Ice Lake, maybe code from that repo could be recycled.

mathstuf, 2019-12-21:
Does this include Comet Lake? Or should I open another issue?

uartie, 2020-06-04:
There are no plans to add support for additional hardware in this driver.  Please use the https://github.com/intel/media-driver (iHD) for all platforms after CFL.

closed 2020-06-04 by uartie