intel/intel-vaapi-driver #470: CFL: Encoding to VP9 with B-frames above 1 results in corrupted output with FFmpeg

Brainiarc7, 2019-05-30:
Hello there,

When encoding with FFmpeg using the `vp9_vaapi` encoder using a B-frame count above 1, regardless of the B-frame reference depth, the resulting video is corrupt.

```sh
#!/bin/bash
#LIBVA_DRIVER_NAME=i965 in this session.

LIBVA_DRIVER_NAME=i965 FFREPORT="level=32:file=vaapi-vp9-bframes.log" ffmpeg -y -stats -qphist -benchmark -vstats_file $(pwd)/vaapi-vp9-bframes.csv \
-vsync 1 -use_wallclock_as_timestamps 1 -fflags +genpts \
-init_hw_device vaapi=va:/dev/dri/renderD128 -filter_hw_device va -hwaccel vaapi \
-i $(pwd)/rv.mkv -vf 'format=vaapi|nv12,hwupload' -bsf:v vp9_raw_reorder,vp9_superframe \
-c:v vp9_vaapi -rc_mode:v VBR -loop_filter_level:v 63 -loop_filter_sharpness:v 15 -b_depth:v 1 -refs:v 16 -b_strategy:v 1 -bf:v 1 \
-b:v 7.5M -minrate:v 7.5M -maxrate:v 15M -bufsize:v 7.5M -r:v 60 -an -f mp4 -y $(pwd)/vaapi-vp9-bframes.mp4
```

**Logs:**

(a). vaapi-vp9-bframes.log

```sh
ffmpeg started on 2019-05-31 at 01:02:29
Report written to "vaapi-vp9-bframes.log"
Command line:
ffmpeg -y -stats -qphist -benchmark -vstats_file /home/brainiarc7/Desktop/src/benchmarks/final/vaapi-vp9-bframes.csv -vsync 1 -use_wallclock_as_timestamps 1 -fflags +genpts -init_hw_device "vaapi=va:/dev/dri/renderD128" -filter_hw_device va -hwaccel vaapi -i /home/brainiarc7/Desktop/src/benchmarks/final/rv.mkv -vf "format=vaapi|nv12,hwupload" -bsf:v vp9_raw_reorder,vp9_superframe -c:v vp9_vaapi -rc_mode:v VBR -loop_filter_level:v 63 -loop_filter_sharpness:v 15 -b_depth:v 1 -refs:v 16 -b_strategy:v 1 -bf:v 1 -b:v 7.5M -minrate:v 7.5M -maxrate:v 15M -bufsize:v 7.5M -r:v 60 -an -f mp4 -y /home/brainiarc7/Desktop/src/benchmarks/final/vaapi-vp9-bframes.mp4
ffmpeg version N-93945-g02333fe394 Copyright (c) 2000-2019 the FFmpeg developers
  built with gcc 7 (Ubuntu 7.4.0-1ubuntu1~18.04)
  configuration: --pkg-config-flags=--static --prefix=/home/brainiarc7 --bindir=/home/brainiarc7/bin --extra-cflags=-I/home/brainiarc7/include --extra-ldflags=-L/home/brainiarc7/lib --enable-cuda-nvcc --enable-cuvid --enable-libnpp --extra-cflags=-I/usr/local/cuda/include/ --extra-ldflags=-L/usr/local/cuda/lib64/ --enable-nvenc --extra-cflags=-I/opt/intel/mediasdk/include --extra-ldflags=-L/opt/intel/mediasdk/lib --extra-ldflags=-L/opt/intel/mediasdk/plugins --enable-libmfx --enable-libass --enable-vaapi --disable-debug --enable-libvorbis --enable-libvpx --enable-libdrm --enable-opencl --enable-gpl --cpu=native --enable-opengl --enable-libfdk-aac --enable-libx264 --enable-libx265 --enable-openssl --extra-libs='-lpthread -lm -lz' --enable-nonfree
  libavutil      56. 28.100 / 56. 28.100
  libavcodec     58. 52.102 / 58. 52.102
  libavformat    58. 27.103 / 58. 27.103
  libavdevice    58.  7.100 / 58.  7.100
  libavfilter     7. 54.101 /  7. 54.101
  libswscale      5.  4.101 /  5.  4.101
  libswresample   3.  4.100 /  3.  4.100
  libpostproc    55.  4.100 / 55.  4.100
Input #0, matroska,webm, from '/home/brainiarc7/Desktop/src/benchmarks/final/rv.mkv':
  Metadata:
    encoder         : libebml v1.3.7 + libmatroska v1.5.0
    creation_time   : 2019-05-09T10:09:00.000000Z
  Duration: 00:05:16.95, start: 1559253749.402000, bitrate: 7889 kb/s
    Stream #0:0: Video: h264 (High), yuv420p(tv, bt470bg/bt709/bt709, progressive), 2560x1080 [SAR 1:1 DAR 64:27], 60 fps, 60 tbr, 1k tbn, 120 tbc (default)
    Metadata:
      BPS-eng         : 7886082
      DURATION-eng    : 00:05:16.950000000
      NUMBER_OF_FRAMES-eng: 19015
      NUMBER_OF_BYTES-eng: 312436712
      _STATISTICS_WRITING_APP-eng: mkvmerge v33.1.0 ('Primrose') 64-bit
      _STATISTICS_WRITING_DATE_UTC-eng: 2019-05-09 10:09:00
      _STATISTICS_TAGS-eng: BPS DURATION NUMBER_OF_FRAMES NUMBER_OF_BYTES
Stream mapping:
  Stream #0:0 -> #0:0 (h264 (native) -> vp9 (vp9_vaapi))
Press [q] to stop, [?] for help
Output #0, mp4, to '/home/brainiarc7/Desktop/src/benchmarks/final/vaapi-vp9-bframes.mp4':
  Metadata:
    encoder         : Lavf58.27.103
    Stream #0:0: Video: vp9 (vp9_vaapi) (Profile 0) (vp09 / 0x39307076), vaapi_vld, 2560x1080 [SAR 1:1 DAR 64:27], q=-1--1, 7500 kb/s, 60 fps, 15360 tbn, 60 tbc (default)
    Metadata:
      BPS-eng         : 7886082
      DURATION-eng    : 00:05:16.950000000
      NUMBER_OF_FRAMES-eng: 19015
      NUMBER_OF_BYTES-eng: 312436712
      _STATISTICS_WRITING_APP-eng: mkvmerge v33.1.0 ('Primrose') 64-bit
      _STATISTICS_WRITING_DATE_UTC-eng: 2019-05-09 10:09:00
      _STATISTICS_TAGS-eng: BPS DURATION NUMBER_OF_FRAMES NUMBER_OF_BYTES
      encoder         : Lavc58.52.102 vp9_vaapi
frame=   28 fps=0.0 q=-0.0 10000000000000000000000000000000size=     512kB time=00:00:00.41 bitrate=10065.6kbits/s dup=1 drop=23 speed=0.83x    
frame=   58 fps= 58 q=-0.0 10000000000000000000000000000000size=     768kB time=00:00:00.91 bitrate=6863.3kbits/s dup=1 drop=51 speed=0.912x    
frame=   88 fps= 58 q=-0.0 20000000000000000000000000000000size=    1280kB time=00:00:01.41 bitrate=7401.6kbits/s dup=1 drop=78 speed=0.94x    
frame=  118 fps= 59 q=-0.0 20000000000000000000000000000000size=    1536kB time=00:00:01.91 bitrate=6565.0kbits/s dup=1 drop=104 speed=0.954x    
[... 361 lines cut ...]
frame= 9921 fps= 60 q=-0.0 80000000000000000000000000000000size=  157184kB time=00:02:45.31 bitrate=7789.0kbits/s dup=1 drop=8374 speed=   1x    
frame= 9950 fps= 60 q=-0.0 80000000000000000000000000000000size=  157440kB time=00:02:45.78 bitrate=7779.7kbits/s dup=1 drop=8404 speed=0.999x    
frame= 9980 fps= 60 q=-0.0 80000000000000000000000000000000size=  157952kB time=00:02:46.28 bitrate=7781.6kbits/s dup=1 drop=8434 speed=0.999x    
[mp4 @ 0x56231a5c5e80] Non-monotonous DTS in output stream 0:0; previous: 2559488, current: 2559488; changing to 2559489. This may result in incorrect timestamps in the output file.
frame=10010 fps= 60 q=-0.0 80000000000000000000000000000000size=  158976kB time=00:02:46.78 bitrate=7808.5kbits/s dup=1 drop=8460 speed=0.999x    
frame=10041 fps= 60 q=-0.0 80000000000000000000000000000000size=  159488kB time=00:02:47.31 bitrate=7808.7kbits/s dup=1 drop=8487 speed=   1x    
frame=10072 fps= 60 q=-0.0 80000000000000000000000000000000size=  159488kB time=00:02:47.81 bitrate=7785.4kbits/s dup=1 drop=8517 speed=   1x    
frame=10102 fps= 60 q=-0.0 80000000000000000000000000000000size=  160000kB time=00:02:48.31 bitrate=7787.2kbits/s dup=1 drop=8547 speed=   1x    
frame=10131 fps= 60 q=-0.0 80000000000000000000000000000000size=  160512kB time=00:02:48.81 bitrate=7789.0kbits/s dup=1 drop=8575 speed=   1x    
frame=10161 fps= 60 q=-0.0 80000000000000000000000000000000size=  161024kB time=00:02:49.31 bitrate=7790.8kbits/s dup=1 drop=8600 speed=0.999x    
frame=10192 fps= 60 q=-0.0 80000000000000000000000000000000size=  161536kB time=00:02:49.81 bitrate=7792.5kbits/s dup=1 drop=8625 speed=0.999x    
frame=10222 fps= 60 q=-0.0 80000000000000000000000000000000size=  161792kB time=00:02:50.31 bitrate=7782.0kbits/s dup=1 drop=8650 speed=0.999x    
[mp4 @ 0x56231a5c5e80] Non-monotonous DTS in output stream 0:0; previous: 2623488, current: 2623488; changing to 2623489. This may result in incorrect timestamps in the output file.
frame=10253 fps= 60 q=-0.0 80000000000000000000000000000000size=  162560kB time=00:02:50.85 bitrate=7794.5kbits/s dup=1 drop=8676 speed=   1x    
frame=10283 fps= 60 q=-0.0 80000000000000000000000000000000size=  163072kB time=00:02:51.35 bitrate=7796.2kbits/s dup=1 drop=8703 speed=   1x    
[matroska,webm @ 0x56231a59e700] Element at 0x12a19b6f ending at 0x12a1a20e exceeds containing master element ending at 0x12a19b69
[mp4 @ 0x56231a5c5e80] Non-monotonous DTS in output stream 0:0; previous: 2635776, current: 2635776; changing to 2635777. This may result in incorrect timestamps in the output file.
frame=10298 fps= 60 q=-0.0 L80000000000000000000000000000000size=  163312kB time=00:02:51.60 bitrate=7796.3kbits/s dup=1 drop=8718 speed=   1x    
video:163266kB audio:0kB subtitle:0kB other streams:0kB global headers:0kB muxing overhead: 0.028138%
bench: utime=73.547s stime=10.931s rtime=171.678s
bench: maxrss=118540kB
```

See the output file [here](https://www.dropbox.com/s/wt05nkqmuwmeu8h/vaapi-vp9-bframes.mp4?dl=0), and see the type of corruption I'm referring to.
Completely disabling B-frames is the only way to get usable output.

Brainiarc7, 2019-06-19:
Any updates on this?

imdark, 2020-11-10:
I am not seeing any corraption in the output file, can you give some better notes on this?