intel/intel-vaapi-driver #472: Heavy Work Let X11 Reset Or GPU Hang

zhoub, 2019-06-03:
Hi !

Still around Pentium N4200.

Example, run a very heavy OpenGL program, **intel_gpu_top** might shows at least **50%** busy.

Then run the **h264encode**, maybe 2 or 3 instances, to encode same video file, so that the **intel_gpu_top** might display the graphics core has a near **90%** busy time.

Sometimes the screen might become black when start a new instance of *h264encode*, then all the programs crash without any warning, and back to the desktop login window.

Really have no idea about this. Just a memo here. Maybe somebody could share some tips to debug the driver or log the system ? Thanks very much.

shirokovoi, 2019-06-03:
Is there something in dmesg.log?

zhoub, 2019-06-03:
[dmesg.log](https://github.com/intel/intel-vaapi-driver/files/3246504/dmesg.log)

It seems that there is an out of memory at the end, but I didn't see the TestOpenGL program has a leak through System Monitor.

Thanks !

zhoub, 2019-06-03:
It seems like there is an memory leak in the decoding program, I would investigate later.

closed 2019-06-03 by zhoub