intel/intel-vaapi-driver #473: VP9 HW encoder generates "marked" video frames if bitrate is changed during encoding

rosetta-jpn, 2019-06-13:
VP9 HW encoder with i965 driver generates such stream that "marked" can be seen in the decoded frames of the bitstream. See attached vp9_hw_enc_KBL.png.
This issue happens only when bitrate is changed during encoding. Further, the bitrate of the generated stream becomes quite unstable.
This issue can be observed in https://appr.tc/?debug=loopback&vsc=vp9&vrc=vp9 with VA-API VP9 encoding.

To help reproducing the issue easily for intel, I create the patch to change bitrate during encoding. You can download yami.tar.gz in https://drive.google.com/file/d/1KtyXkOW-DHW5_fW4tj_GuQGb08_TgWmG.
It also have frames-420.yuv, raw 900 YUV frames (I420). You can try the following command to reproduce the issue.

$ ./yamiencode -i frames-420.yuv -s I420 -W 640 -H 360 -c VP9 -o out.vp9.ivf
[image]

rosetta-jpn, 2019-06-13:
@sreerenjb @xhaihao 

haihao, could you take a look?

santoshavati, 2019-07-02:
@rosetta-jpn @ I'll be working on to reproduce the issue on kabylake platform. I have few question to understand issue, 
1. I have downloaded yami.tar.gz and extracted, one of file  "0001-Bitrate-change-patch.patch" thought it's is patch file but its 142MB file looks like some executable. can please let me know what is this file and how it can be used in chromeOS.
2. What is "marked" in .png which you attached. can please send more details what is exact issue here.

86423355844265459587182778, 2019-07-07:
@santoshavati 
The file "0001-Bitrate-change-patch.patch" appears to be a .tar.gz file. If you open it using a hex editor, you will see that the file starts with `1F 8B`. It contains one file (`yami/frames/frames-420.yuv`, 296.6 MiB uncompressed) only.

sreerenjb, 2019-08-02:
@rosetta-jpn as @santoshavati  mentioned, the included patch is not actually a patch :) Could you please provide the libyami patch ?

rosetta-jpn, 2019-08-15:
Hi, Sorry for delay.
I uploaded with a correct patch again. Please take a look.
https://drive.google.com/file/d/1j1qRUIamNt56d-FJyCPfImkVsI5d8ozi

Thanks.

sreerenjb, 2019-09-19:
@rosetta-jpn Please let me know if https://github.com/intel/intel-vaapi-driver/pull/486 is fixes the issue. I will upload epatch to chrome too.

rosetta-jpn, 2019-10-16:
After #486, the symptom becomes better, but it doesn't fix this issue perfectly.
When something is moved in a captured content, the encoded stream becomes blocky around of the moving thing.