intel/intel-vaapi-driver #475: Requested 'libva >= 1.4.0' but version of libva is 0.38.1

JustingLYX, 2019-07-08:
I want to install libva,intel-vaapi-driver and libva-utils. First I installed libva like this:

Git clone https://github.com/01org/libva
Cd libva
./autogen.sh
./configure
Make
Make install
And libva version: libva - 2.5.0.pre1 (VA-API 1.5.0)
[image]

But when I try to install intel-vaapi-driver,like this:
git clone https://github.com/01org/intel-vaapi-driver
cd intel-vaapi-driver
./autogen.sh
./configure
make
make install

 it gives the following error:
Configure: error: Package requirements (libva >= 1.4.0) were not met:
Requested 'libva >= 1.4.0' but version of libva is 0.38.1
[image]

0987363, 2019-07-31:
+1

the 0.38.1 is VA-API version, not libva version

zhoub, 2019-12-02:
Are you using system's vainfo ?