intel/intel-vaapi-driver #502: iris_drv_video.so not exsit

12101111, 2020-03-07:
I'm using mesa20 and new "iris" driver for intel gen8+ graphic card(In my case is HD 530).

libva try to load `iris_drv_video.so` when mesa using iris driver, but it don't exsit .
A workaround is create a symbol link to `i965_drv_video.so`.

```log
$ vainfo
libva info: VA-API version 1.6.0
libva info: Trying to open /usr/lib/va/drivers/iris_drv_video.so
libva info: va_openDriver() returns -1
vaInitialize failed with error code -1 (unknown libva error),exit
$ cd /usr/lib/va/drivers
$ ls
i965_drv_video.so
$ sudo ln -s i965_drv_video.so iris_drv_video.so
$ ls
i965_drv_video.so  iris_drv_video.so
$ vainfo
libva info: VA-API version 1.6.0
libva info: Trying to open /usr/lib/va/drivers/iris_drv_video.so
libva info: Found init function __vaDriverInit_1_6
libva info: va_openDriver() returns 0
vainfo: VA-API version: 1.6 (libva 2.6.1)
vainfo: Driver version: Intel i965 driver for Intel(R) Skylake - 2.4.0.pre1 (2.4.0.pre1)
vainfo: Supported profile and entrypoints
      VAProfileMPEG2Simple            :	VAEntrypointVLD
      VAProfileMPEG2Simple            :	VAEntrypointEncSlice
      VAProfileMPEG2Main              :	VAEntrypointVLD
      VAProfileMPEG2Main              :	VAEntrypointEncSlice
      VAProfileH264ConstrainedBaseline:	VAEntrypointVLD
      VAProfileH264ConstrainedBaseline:	VAEntrypointEncSlice
      VAProfileH264ConstrainedBaseline:	VAEntrypointEncSliceLP
      VAProfileH264ConstrainedBaseline:	VAEntrypointFEI
      VAProfileH264ConstrainedBaseline:	VAEntrypointStats
      VAProfileH264Main               :	VAEntrypointVLD
      VAProfileH264Main               :	VAEntrypointEncSlice
      VAProfileH264Main               :	VAEntrypointEncSliceLP
.....
```

uartie, 2020-03-07:
The iris driver is part of the mesa project... so, this sounds like a mesa bug.  Please file a report with them.

This project is for the i965 driver.

uartie, 2020-03-07:
Perhaps look at https://gitlab.freedesktop.org/mesa/mesa/-/merge_requests/283

uartie, 2020-03-07:
> This project is for the i965 driver.

This project is for the i965 **video** driver.  It is not a opengl driver.

closed 2020-03-07 by uartie

uartie, 2020-03-27:
See intel/libva#396 for follow up

uartie, 2020-04-14:
FYI, https://github.com/intel/libva/pull/406