intel/intel-vaapi-driver #510: Please make a gcc 10.1.0 compatible release or pre-release available

Flexximilian, 2020-05-19:
Since gcc 10.1.0 was officially released on May, 7th 2020, it would be truly helpful to have an official (pre-)release tag that builds on it. #503 and #509 were fixed, but a pre-release will be needed to get at least ~amd64 (unstable) support on Gentoo, and likely other distributions.

Thank you for your great work!

uartie, 2020-05-20:
@xhaihao can you do a maintenance release?

Flexximilian, 2020-05-20:
Just so I don't have to ask you twice if any other error would pop up, I've tried merging a live ebuild of commit 625d2651258db881d92eb5ffc97cb4f4f1fda239 (current master/HEAD) on my up-to-date ~amd64 Gentoo with gcc 10.1.0, and it built just fine. That's great news for a quick adoption in the main repo, and that there aren't any more gcc 10.1.0 issues (using a fresh toolchain) as far as I can tell.

Once you have a release out, I'll test it once more and issue an ebuild version bump request to the Gentoo Bugzilla.

If there's anything more I can help with, let me know.

xhaihao, 2020-05-22:
@uartie @Flexximilian Thanks for reminding me. We will cut a new release soon.

rubyFeedback, 2020-06-02:
GCC 10.1.0 causes quite some issues in general; not just here but elsewhere too, not only in gentoo. Not really having much to contribute here :) but I just wanted to mention it; even the ruby core team had to report a change upstream. Gues they changed a lot from 9 -> 10 in GCC.

Flexximilian, 2020-06-06:
No, not really. One flag's default was flipped (not by accident and after a long discussion about it). Gcc 10 sets -fno-common by default, and that made very long standing bugs (which were handled gracefully due to gcc-9's default of -fcommon) now cause compile errors. In other words: gcc got stricter by default. Now you can explicitly set -fcommon to get the old behaviour, or fix the bugs (multiple definitions of the same identifier, basically), which is obviously the preferred way of dealing with this. A lot of packages actually have been fixed already, but often no new release was made yet, since those fixes are made on the bleeding edge of the codebase.

xhaihao, 2020-06-08:
new version of intel-vaapi-driver: https://github.com/intel/intel-vaapi-driver/releases/tag/2.4.1

closed 2020-06-08 by xhaihao