intel/intel-vaapi-driver #512: [Small change] Would it be possible to release a "full" 2.4.1 tarball?

rubyFeedback, 2020-06-02:
Hey guys,

The last release had this URL:

https://github.com/intel/intel-vaapi-driver/releases/download/2.4.0/intel-vaapi-driver-2.4.0.tar.bz2

There is currently no 2.4.1 release:

https://github.com/intel/intel-vaapi-driver/releases/download/2.4.1/intel-vaapi-driver-2.4.1.tar.bz2
^^^ wget fails

I know we can download it here instead:

https://github.com/intel/intel-vaapi-driver/archive/2.4.1.tar.gz

And this works fine. But I also govern downloads via a script, and while the script can parse
the above URL too, I much prefer the longer name such as "intel-vaapi-driver-2.4.1.tar.bz2"

The last URL for 2.4.0 works fine, so I assume that this was perhaps forgotten. That is why
I added this issue request - hope it is not too much work to change (I don't know how to
change it myself, have no real experience with the github interface, aside from using it for
issue discussions, sorry).

Anyway, keep up the good work - cheers!

Flexximilian, 2020-06-06:
This relates/applies to #510 also. The 2.4.1 tarball compiles fine on Gentoo with gcc 10, BTW.

xhaihao, 2020-06-08:
@Flexximilian I will upload the tar ball, thanks for reminding me.

xhaihao, 2020-06-08:
Done, please try with https://github.com/intel/intel-vaapi-driver/releases/download/2.4.1/intel-vaapi-driver-2.4.1.tar.bz2

closed 2020-06-08 by xhaihao