intel/intel-vaapi-driver #513: VP9 and HEVC accelerated decode support on Haswell and Broadwell

ryao, 2020-06-05:
https://forums.intel.com/s/question/0D50P0000490MoXSAU/intel-iris-and-hd-graphics-driver-update-posted-for-haswell-and-broadwell-version-1536144080?language=en_US

The changelog indicates that the Windows graphics driver supports "partial hardware" accelerated decode of VP9 and "GPU accelerated decode of HEVC" on Haswell and Broadwell. vainfo reports no support for either of the two on my Xeon E3-1276v3 Haswell CPU.

Could we get accelerated decoding support in the Linux driver? I imagine that it could be done vai a compute shader, which is likely what "GPU accelerated decode of HEVC" means.

cshei, 2020-09-02:
Isn't that what https://github.com/intel/intel-hybrid-driver is for? (at least, for VP9)

rathann, 2021-03-10:
Indeed it is:
```
$ vainfo
libva info: VA-API version 1.9.0
libva info: User environment variable requested driver 'i965'
libva info: Trying to open /usr/lib64/dri/i965_drv_video.so
libva info: Found init function __vaDriverInit_1_8
failed to open /usr/lib64/dri/hybrid_drv_video.so
Not using hybrid_drv_video.so
libva info: va_openDriver() returns 0
vainfo: VA-API version: 1.9 (libva 2.9.0)
vainfo: Driver version: Intel i965 driver for Intel(R) Haswell Mobile - 2.4.1
vainfo: Supported profile and entrypoints
      VAProfileMPEG2Simple            :	VAEntrypointVLD
      VAProfileMPEG2Simple            :	VAEntrypointEncSlice
      VAProfileMPEG2Main              :	VAEntrypointVLD
      VAProfileMPEG2Main              :	VAEntrypointEncSlice
      VAProfileH264ConstrainedBaseline:	VAEntrypointVLD
      VAProfileH264ConstrainedBaseline:	VAEntrypointEncSlice
      VAProfileH264Main               :	VAEntrypointVLD
      VAProfileH264Main               :	VAEntrypointEncSlice
      VAProfileH264High               :	VAEntrypointVLD
      VAProfileH264High               :	VAEntrypointEncSlice
      VAProfileH264MultiviewHigh      :	VAEntrypointVLD
      VAProfileH264MultiviewHigh      :	VAEntrypointEncSlice
      VAProfileH264StereoHigh         :	VAEntrypointVLD
      VAProfileH264StereoHigh         :	VAEntrypointEncSlice
      VAProfileVC1Simple              :	VAEntrypointVLD
      VAProfileVC1Main                :	VAEntrypointVLD
      VAProfileVC1Advanced            :	VAEntrypointVLD
      VAProfileNone                   :	VAEntrypointVideoProc
      VAProfileJPEGBaseline           :	VAEntrypointVLD
$ sudo dnf install libva-intel-hybrid-driver
...
$ vainfo
error: can't connect to X server!
libva info: VA-API version 1.9.0
libva info: User environment variable requested driver 'i965'
libva info: Trying to open /usr/lib64/dri/i965_drv_video.so
libva info: Found init function __vaDriverInit_1_8
libva info: va_openDriver() returns 0
vainfo: VA-API version: 1.9 (libva 2.9.0)
vainfo: Driver version: Intel i965 driver for Intel(R) Haswell Mobile - 2.4.1
vainfo: Supported profile and entrypoints
      VAProfileMPEG2Simple            :	VAEntrypointVLD
      VAProfileMPEG2Simple            :	VAEntrypointEncSlice
      VAProfileMPEG2Main              :	VAEntrypointVLD
      VAProfileMPEG2Main              :	VAEntrypointEncSlice
      VAProfileH264ConstrainedBaseline:	VAEntrypointVLD
      VAProfileH264ConstrainedBaseline:	VAEntrypointEncSlice
      VAProfileH264Main               :	VAEntrypointVLD
      VAProfileH264Main               :	VAEntrypointEncSlice
      VAProfileH264High               :	VAEntrypointVLD
      VAProfileH264High               :	VAEntrypointEncSlice
      VAProfileH264MultiviewHigh      :	VAEntrypointVLD
      VAProfileH264MultiviewHigh      :	VAEntrypointEncSlice
      VAProfileH264StereoHigh         :	VAEntrypointVLD
      VAProfileH264StereoHigh         :	VAEntrypointEncSlice
      VAProfileVC1Simple              :	VAEntrypointVLD
      VAProfileVC1Main                :	VAEntrypointVLD
      VAProfileVC1Advanced            :	VAEntrypointVLD
      VAProfileNone                   :	VAEntrypointVideoProc
      VAProfileJPEGBaseline           :	VAEntrypointVLD
      VAProfileVP9Profile0            :	VAEntrypointVLD
$
```
Note the presence of `VAProfileVP9Profile0` in the second `vainfo` output.

Ristovski, 2021-11-26:
It appears that HEVC decode support is still not present, any news/progress?

sakshatshinde, 2024-06-26:
Any progress?