intel/intel-vaapi-driver #516: vaGetImage() doesn't fill the last column and row in chroma plane if the dimension is odd

rosetta-jpn, 2020-07-31:
The title describes the issue.
I found this bug when I map an odd dimension frame with VA Image by vaCreateImage() and vaGetImage().
If I change the part of code to vaDeriveImage(), the issue was gone.

sreerenjb, 2020-08-11:
Any Sample code to share? Is there a corresponding chromeos bug open somewhere?
@xhaihao  ping

sreerenjb, 2020-08-11:
Reproducible with gst:
gst-launch-1.0 -v videotestsrc ! videoscale ! video/x-raw, width=321, height=241 ! vaapipostproc ! xvimagesink

xhaihao, 2020-08-12:
@sreerenjb @rosetta-jpn Could you give a try with https://github.com/xhaihao/intel-vaapi-driver/tree/fix-odd-resolution on your side? It works on my SKL.

sreerenjb, 2020-08-12:
@xhaihao 
Yup, it is fixing the GStreamer pipeline for sure. Tested on KBL.

sreerenjb, 2020-08-12:
Thank you Haihao

sreerenjb, 2020-08-13:
@xhaihao HIro confirmed that it fixed the chrome issue too.
Can you please merge the fix?

rosetta-jpn, 2020-08-18:
Haihao, gentle ping.

xhaihao, 2020-08-18:
@sreerenjb @rosetta-jpn Thanks for testing the patch, I will submit a PR ASAP.

closed 2020-08-19 by xhaihao