intel/intel-vaapi-driver #520: Package requirements (libdrm >= 2.4.52 libdrm_intel) were not met

LinuxOpa, 2020-09-23:
Ubuntu 20.04.1

lshw -short:
display        Xeon E3-1200 v2/3rd Gen Core processor Graphics Controller

autogen.sh:
configure: error: Package requirements (libdrm >= 2.4.52 libdrm_intel) were not met:
No package 'libdrm' found
No package 'libdrm_intel' found

apt search libdrm_intel:
libdrm-intel**1**/focal,now 2.4.101-2 amd64  [Installiert,automatisch]
Benutzerbereichsschnittstelle zu Intel-spezifischen Kernel-DRM-Diensten – Laufzeit

apt show libdrm-intel**1**:
Package: libdrm-intel**1**
Version: 2.4.101-2
Priority: optional
Section: libs
Source: libdrm
Origin: Ubuntu
Maintainer: Ubuntu Developers <ubuntu-devel-discuss@lists.ubuntu.com>
Original-Maintainer: Debian X Strike Force <debian-x@lists.debian.org>
Bugs: https://bugs.launchpad.net/ubuntu/+filebug
Installed-Size: 189 kB
Depends: libc6 (>= 2.17), libdrm2 (>= 2.4.38), libpciaccess0
Homepage: https://cgit.freedesktop.org/mesa/drm/
Task: ubuntu-desktop-minimal, ubuntu-desktop, kubuntu-desktop, xubuntu-core, xubuntu-desktop, lubuntu-desktop, ubuntustudio-desktop-core, ubuntustudio-desktop, ubuntukylin-desktop, ubuntu-mate-core, ubuntu-mate-desktop, ubuntu-budgie-desktop
Download-Size: 60,2 kB
APT-Manual-Installed: no
APT-Sources: http://archive.ubuntu.com/ubuntu focal/main amd64 Packages
Description: Benutzerbereichsschnittstelle zu Intel-spezifischen Kernel-DRM-Diensten – Laufzeit
 Diese Bibliothek implementiert die Userspace-Schnittstelle zu den Kernel-
 DRM-Diensten. DRM steht für »Direct Rendering Manager«, der den Kernel-
 Teil der »Direct Rendering Infrastructure« (DRI) darstellt. DRI wird zur
 Zeit unter Linux eingesetzt, um hardwarebeschleunigte OpenGL-Treiber zur
 Verfügung zu stellen.

uartie, 2020-09-23:
Did you install libdrm-dev?

LinuxOpa, 2020-09-23:
that's it.
thanks

closed 2020-09-23 by LinuxOpa