intel/intel-vaapi-driver #523: libva error: /usr/lib/x86_64-linux-gnu/dri/i965_drv_video.so init failed

bjoernhohe, 2020-11-24:
I have trouble using GPU acceleration in different browser (Brave, Chromium, Firefox none works)
When starting for example Brave Browser the following appears in the log:
```
libva error: /usr/lib/x86_64-linux-gnu/dri/i965_drv_video.so init failed
[63863:63863:1124/202012.764559:ERROR:vaapi_wrapper.cc(541)] vaInitialize failed: unknown libva error
[63863:63863:1124/202012.769317:ERROR:sandbox_linux.cc(374)] InitializeSandbox() called with multiple threads in process gpu-process.
[63834:63834:1124/202013.062974:ERROR:sharing_service.cc(218)] Device registration failed with fatal error

```
vainfo reports the following:
```
libva info: VA-API version 1.8.0
libva info: Trying to open /usr/lib/x86_64-linux-gnu/dri/iHD_drv_video.so
libva info: Found init function __vaDriverInit_1_8
libva info: va_openDriver() returns 0
vainfo: VA-API version: 1.8 (libva 2.8.0)
vainfo: Driver version: Intel iHD driver for Intel(R) Gen Graphics - 20.2.0 ()
vainfo: Supported profile and entrypoints
      VAProfileNone                   :	VAEntrypointVideoProc
      VAProfileNone                   :	VAEntrypointStats
      VAProfileMPEG2Simple            :	VAEntrypointVLD
      VAProfileMPEG2Simple            :	VAEntrypointEncSlice
      VAProfileMPEG2Main              :	VAEntrypointVLD
      VAProfileMPEG2Main              :	VAEntrypointEncSlice
      VAProfileH264Main               :	VAEntrypointVLD
      VAProfileH264Main               :	VAEntrypointEncSlice
      VAProfileH264Main               :	VAEntrypointFEI
      VAProfileH264Main               :	VAEntrypointEncSliceLP
      VAProfileH264High               :	VAEntrypointVLD
      VAProfileH264High               :	VAEntrypointEncSlice
      VAProfileH264High               :	VAEntrypointFEI
      VAProfileH264High               :	VAEntrypointEncSliceLP
      VAProfileVC1Simple              :	VAEntrypointVLD
      VAProfileVC1Main                :	VAEntrypointVLD
      VAProfileVC1Advanced            :	VAEntrypointVLD
      VAProfileJPEGBaseline           :	VAEntrypointVLD
      VAProfileJPEGBaseline           :	VAEntrypointEncPicture
      VAProfileH264ConstrainedBaseline:	VAEntrypointVLD
      VAProfileH264ConstrainedBaseline:	VAEntrypointEncSlice
      VAProfileH264ConstrainedBaseline:	VAEntrypointFEI
      VAProfileH264ConstrainedBaseline:	VAEntrypointEncSliceLP
      VAProfileVP8Version0_3          :	VAEntrypointVLD
      VAProfileVP8Version0_3          :	VAEntrypointEncSlice
      VAProfileHEVCMain               :	VAEntrypointVLD
      VAProfileHEVCMain               :	VAEntrypointEncSlice
      VAProfileHEVCMain               :	VAEntrypointFEI
      VAProfileHEVCMain               :	VAEntrypointEncSliceLP
      VAProfileHEVCMain10             :	VAEntrypointVLD
      VAProfileHEVCMain10             :	VAEntrypointEncSlice
      VAProfileHEVCMain10             :	VAEntrypointEncSliceLP
      VAProfileVP9Profile0            :	VAEntrypointVLD
      VAProfileVP9Profile1            :	VAEntrypointVLD
      VAProfileVP9Profile2            :	VAEntrypointVLD
      VAProfileVP9Profile3            :	VAEntrypointVLD
      VAProfileHEVCMain422_10         :	VAEntrypointVLD
      VAProfileHEVCMain422_10         :	VAEntrypointEncSlice
      VAProfileHEVCMain444            :	VAEntrypointVLD
      VAProfileHEVCMain444            :	VAEntrypointEncSliceLP
      VAProfileHEVCMain444_10         :	VAEntrypointVLD
      VAProfileHEVCMain444_10         :	VAEntrypointEncSliceLP
```

```
lspci | grep VGA
00:02.0 VGA compatible controller: Intel Corporation Iris Plus Graphics G7 (rev 07)
```

My laptop is a Razer Blade Stealt 2020 also with an NVIDIA 1650TI (but error also occurs when this one is deactivated) with POP_OS 20.100
GPU acceleration works in Kdenlive or VLC for both GPU

bjoernhohe, 2020-11-26:
got it working in firefox by setting security.sandbox.content.level to 0 instead of 4 

https://bugzilla.mozilla.org/show_bug.cgi?id=1619585

closed 2020-11-26 by bjoernhohe

mizzunet, 2021-12-02:
@bjoernhohe what about chromium based browsers? you have any workaround to share? I also have same error with mpv running with vaapi flags.

mizzunet, 2021-12-02:
Also I do not seem to have an output like you have,
```
gst-inspect-1.0 vaapi
No such element or plugin 'vaapi'
```

bjoernhohe, 2021-12-04:
tried the patched chromium, official chrome, brave...no luck

bjoernhohe, 2021-12-04:
did you install the intel drivers?

mizzunet, 2021-12-04:
Yeah, mine is Haswell. I have installed `libva-intel-driver `
```
 ~  vainfo
vainfo: VA-API version: 1.13 (libva 2.13.0)
vainfo: Driver version: Intel i965 driver for Intel(R) Haswell Mobile - 2.4.1
vainfo: Supported profile and entrypoints
      VAProfileMPEG2Simple            :	VAEntrypointVLD
      VAProfileMPEG2Simple            :	VAEntrypointEncSlice
      VAProfileMPEG2Main              :	VAEntrypointVLD
      VAProfileMPEG2Main              :	VAEntrypointEncSlice
      VAProfileH264ConstrainedBaseline:	VAEntrypointVLD
      VAProfileH264ConstrainedBaseline:	VAEntrypointEncSlice
      VAProfileH264Main               :	VAEntrypointVLD
      VAProfileH264Main               :	VAEntrypointEncSlice
      VAProfileH264High               :	VAEntrypointVLD
      VAProfileH264High               :	VAEntrypointEncSlice
      VAProfileH264MultiviewHigh      :	VAEntrypointVLD
      VAProfileH264MultiviewHigh      :	VAEntrypointEncSlice
      VAProfileH264StereoHigh         :	VAEntrypointVLD
      VAProfileH264StereoHigh         :	VAEntrypointEncSlice
      VAProfileVC1Simple              :	VAEntrypointVLD
      VAProfileVC1Main                :	VAEntrypointVLD
      VAProfileVC1Advanced            :	VAEntrypointVLD
      VAProfileNone                   :	VAEntrypointVideoProc
      VAProfileJPEGBaseline           :	VAEntrypointVLD
```

mizzunet, 2021-12-04:
After reinstallation on Arch, I got ouput for 
```
~  gst-inspect-1.0 vaapi
Plugin Details:
  Name                     vaapi
  Description              VA-API based elements
  Filename                 /usr/lib/gstreamer-1.0/libgstvaapi.so
  Version                  1.18.5
  License                  LGPL
  Source module            gstreamer-vaapi
  Source release date      2021-09-08
  Binary package           gstreamer-vaapi
  Origin URL               https://www.archlinux.org/

  vaapijpegdec: VA-API JPEG decoder
  vaapimpeg2dec: VA-API MPEG2 decoder
  vaapih264dec: VA-API H264 decoder
  vaapivc1dec: VA-API VC1 decoder
  vaapipostproc: VA-API video postprocessing
  vaapidecodebin: VA-API Decode Bin
  vaapisink: VA-API sink
  vaapimpeg2enc: VA-API MPEG-2 encoder
  vaapih264enc: VA-API H264 encoder

  9 features:
  +-- 9 elements

```

bjoernhohe, 2021-12-05:
You could try in Hardware acceleration works in Apps live VLC