intel/intel-vaapi-driver #525: Consider merging ChromeOS "Avoid GPU crash with malformed streams" patch

yellowdoge, 2020-12-22:
During ChromeOS intel-vaapi-driver uprev to 2.4 (https://chromium-review.googlesource.com/c/chromiumos/overlays/chromiumos-overlay/+/2597004) we found this patch that hasn't been upstreamed: https://source.chromium.org/chromiumos/chromiumos/codesearch/+/main:src/third_party/chromiumos-overlay/x11-libs/libva-intel-driver/files/Avoid-GPU-crash-with-malformed-streams.patch?q=Avoid-GPU-crash-with-malformed-streams.patch .

Basically it makes sure that P-frames with no reference frames (due to malformed bitstream) are handled appropriately.

yellowdoge, 2020-12-22:
https://b.corp.google.com/issues/176128852 is the partner ChromeOS issue.

xhaihao, 2021-09-17:
fixed, see https://github.com/intel/intel-vaapi-driver/pull/537

closed 2021-09-17 by xhaihao