intel/intel-vaapi-driver #528: parole: /usr/lib/dri/iHD_drv_video.so init failed - Unable to play videos

MartinX3, 2021-01-12:
Intel Skylake 6700HQ with Nvidia GTX 940M
Arch Linux / EndeavourOS
intel-media-sdk 20.5.1
intel-media-driver 20.4.5
libva 2.10.0
```
$ sudo cat /sys/kernel/debug/dri/128/name
i915 dev=0000:00:02.0 master=pci:0000:00:02.0 unique=0000:00:02.0
$ sudo cat /sys/kernel/debug/dri/129/name
nvidia-drm dev=0000:02:00.0 unique=0000:02:00.0
```

```
$ GST_DEBUG=2 parole '/run/media/martin/extLinux/simplescreenrecorder-2020-11-24_18.17.54.mp4' 
0:00:00.021679755 61762 0x55d4d8d2a290 WARN     GST_ELEMENT_FACTORY gstelementfactory.c:467:gst_element_factory_make: no such element factory "autoimagesink"!
0:00:00.309384106 61762 0x55d4d8d2a290 WARN                 basesrc gstbasesrc.c:3688:gst_base_src_start_complete:<source> pad not activated yet
0:00:00.309617733 61762 0x55d4d8d2a290 WARN                 basesrc gstbasesrc.c:3688:gst_base_src_start_complete:<source> pad not activated yet
0:00:00.339973354 61762 0x55d4d95565e0 WARN                 qtdemux qtdemux_types.c:244:qtdemux_type_get: unknown QuickTime node type pasp
0:00:00.340034060 61762 0x55d4d95565e0 WARN                 qtdemux qtdemux.c:3066:qtdemux_parse_trex:<qtdemux0> failed to find fragment defaults for stream 1
0:00:00.340923386 61762 0x55d4d95565e0 WARN                 qtdemux qtdemux.c:3066:qtdemux_parse_trex:<qtdemux0> failed to find fragment defaults for stream 2
DRM_IOCTL_I915_GEM_APERTURE failed: Invalid argument
Assuming 131072kB available aperture size.
May lead to reduced performance or incorrect rendering.
get chip id failed: -1 [22]
param: 4, val: 0
0:00:00.354372669 61762 0x55d4d95565e0 WARN                   vaapi gstvaapiutils.c:78:gst_vaapi_warning: /usr/lib/dri/iHD_drv_video.so init failed
0:00:00.506574788 61762 0x55d4d959f300 ERROR          vaapipostproc gstvaapipostproc.c:1087:gst_vaapipostproc_process_vpp:<vaapipostproc0> failed to apply VPP filters (error 2)
0:00:00.520345335 61762 0x55d4d95565e0 WARN                 qtdemux qtdemux.c:6545:gst_qtdemux_loop:<qtdemux0> error: Internal data stream error.
0:00:00.520363716 61762 0x55d4d95565e0 WARN                 qtdemux qtdemux.c:6545:gst_qtdemux_loop:<qtdemux0> error: streaming stopped, reason error (-5)
```

uartie, 2021-01-27:
@MartinX3 this repository is for i965 vaapi driver...

@xinfengzhang... please transfer to https://github.com/intel/media-driver

MartinX3, 2021-01-27:
Manually transferred...
https://github.com/intel/media-driver/issues/1131

closed 2021-01-27 by MartinX3