intel/intel-vaapi-driver #533: Crocus driver ?

thetick1234567890, 2021-07-22:
Are there plans for a crocus vaapi driver ?  When I switch mesa to crocus driver I get an error with vainfo that states  crocus_drv_video.so is not found.  If make a symbolic link : "sudo ln -s crocus_drv_video.so i965_drv_video.so" hardware acceeration works fine as the current i965 is used.

lorn10, 2021-08-04:
I can confirm this problem for an Intel Sandy Bridge based system. The VAAPI support is currently missing when the new Gallium 3D Crocus driver is used.

However, there is a comment at phoronix forums from an Intel developer which notes that "Media support would certainly be neat" But it seems somewhat unclear when this will happen. So as for now, we only can wait and hope that this will happen sometimes in the near future. Source: [Mesa's New "Crocus" OpenGL Driver Is Performing Well For Old Intel Hardware](https://www.phoronix.com/forums/forum/linux-graphics-x-org-drivers/opengl-vulkan-mesa-gallium3d/1262210-mesa-s-new-crocus-opengl-driver-is-performing-well-for-old-intel-hardware/page2)

jbeich, 2021-11-27:
Can you reproduce with libva from `master` branch? If not close the issue.

lorn10, 2021-11-30:
This doesn't land in Mesa, right? I am on `Mesa 22.0.0-devel (git-b5848b2 2021-11-27 focal-oibaf-ppa)` and I still get an error message when a h264 video is played via mpv and crocus is in use.

Maybe @thetick1234567890 is able to build `libva` from `master` an try this out.

thetick1234567890, 2021-12-31:
Thanks! Can you please release a new official package as I'm hoping this gets in the 22.04 Ubuntu release.

closed 2021-12-31 by thetick1234567890