intel/intel-vaapi-driver #536: Driver consistently fails with - vaEndPicture invalid parameter

evelikov, 2021-09-07:
Greetings team,

For some H264 decoding workloads the driver consistently fails with the mentioned "vaEndPicture (va_display, context) failed: invalid parameter (18)". Either of the following triggers the issue - [1](https://gitlab.steamos.cloud/steamrt/steam-runtime-tools/-/blob/master/helpers/check-va-api.c) and [2](https://github.com/intel/libva-utils/blob/master/vendor/intel/avcstreamoutdemo.c). Note that for the latter may fail with `main:vaCreateBuffer (1379) failed,exit` - if so comment out the respective code section.

Casual look at the driver suggests that the stream data is invalid, although I did not narrow it down any further.

Any tips and suggestions are appreciated \o/

Affected hardware:
 - Broadwell - i7-5500U aka Intel Corporation HD Graphics 5500 aka 8086:1616
 - Skylake - Core i5 6600K (I think can check next week)

evelikov, 2021-11-01:
Humble poke?

evelikov, 2021-11-23:
@uartie @xhaihao any pointers?

ilikenwf, 2024-10-14:
I think this is similar to the decode issues I see for some HDR av1 and vp9 videos using an arc card with kodi's youtube addon...

https://github.com/xbmc/xbmc/issues/23957