intel/intel-vaapi-driver #539: HW VP9 encode GPU HANG on i915/RKL - Resetting vcs0 for preemption time out - GPU HANG: ecode 12:4:28ffff7d, in ffmpeg

ajgnet, 2021-09-17:
Unable to HW encode VP9 on Rocket Lake i7-11700K. Tried various kernels, applications, intel-vaapi-driver versions.

`ffmpeg -vaapi_device /dev/dri/renderD128 -hwaccel vaapi -hwaccel_output_format vaapi -I "input.mp4" -c:v vp9_vaapi output.mp4
`

Encoding fails ~50% of the time with a GPU hang.

`[ 3806.462671] i915 0000:00:02.0: [drm] Resetting vcs0 for preemption time out
[ 3806.462689] i915 0000:00:02.0: [drm] ffmpeg[1377] context reset due to GPU hang
[ 3806.463476] i915 0000:00:02.0: [drm] GPU HANG: ecode 12:4:28ffff7d, in ffmpeg [1377]`

HEVC encoding works as expected
`ffmpeg -vaapi_device /dev/dri/renderD128 -hwaccel vaapi -hwaccel_output_format vaapi -I "input.mp4" -c:v hevc_vaapi output.mp4
`

ajgnet, 2021-10-02:
Just a quick update to the above bug, this appears to only occur with videos longer than 10 seconds in length when converting to VP9 only. Perhaps it is a memory bug of some kind. Happy to provide any more details as requested that may be helpful.

dalcombright, 2021-12-28:
> Just a quick update to the above bug, this appears to only occur with videos longer than 10 seconds in length when converting to VP9 only. Perhaps it is a memory bug of some kind. Happy to provide any more details as requested that may be helpful.

any luck? I am seeing the same thing using unmanic on unraid. Nothing will convert.

ajgnet, 2022-01-16:
> 

Sadly no. We dropped the processor and went with Alder Lake where everything works as expected.

edwarddavidbaker, 2022-01-16:
I'm not sure which Linux distribution you are using, but you will want to discuss Rocket Lake at https://github.com/intel/media-driver . Also for VP9 encode it might be important to double check loading microcontroller firmware GuC and HuC.