intel/intel-vaapi-driver #545: get chip id failed: -1 [2]

Igortorrente, 2022-02-07:
Hi,

I was debugging a issue related to the vaapi on a radeon gpu. And I decided to compare the behavior with an Intel gpu.
To my surprise, the intel vaapi wasn't working at all.
I couldn't find a solution anywhere, so I'm opening this issue.

```
$ LIBVA_DRIVER_NAME=i965 vainfo
libva info: VA-API version 1.10.0
libva info: User environment variable requested driver 'i965'
libva info: Trying to open /usr/lib/x86_64-linux-gnu/dri/i965_drv_video.so
libva info: Found init function __vaDriverInit_1_8
DRM_IOCTL_I915_GEM_APERTURE failed: Invalid argument
Assuming 131072kB available aperture size.
May lead to reduced performance or incorrect rendering.
get chip id failed: -1 [2]
param: 4, val: 0
libva error: /usr/lib/x86_64-linux-gnu/dri/i965_drv_video.so init failed
libva info: va_openDriver() returns -1
vaInitialize failed with error code -1 (unknown libva error),exit
```
#### Additional info:
Vaapi driver: [i965-va-driver (2.4.1+dfsg1-1)](https://packages.debian.org/bullseye/i965-va-driver)
Model name:  Intel(R) Core(TM) i5-4440 CPU @ 3.10GHz
```
$ lsmod | grep i915
i915                 2768896  3
i2c_algo_bit           16384  3 amdgpu,radeon,i915
ttm                    77824  4 amdgpu,radeon,drm_ttm_helper,i915
drm_kms_helper        303104  3 amdgpu,radeon,i915
cec                    61440  2 drm_kms_helper,i915
drm                   561152  26 gpu_sched,drm_kms_helper,amdgpu,radeon,drm_ttm_helper,i915,ttm
video                  53248  1 i915
```
```
$ uname -a
Linux debian-desktop 5.15.11-xanmod1 #0~git20211222.96a4b9e SMP Wed Dec 22 18:01:17 UTC 2021 x86_64 GNU/Linux
```

Brainiarc7, 2022-12-08:
You most likely have two VAAPI-capable GPUs on that system.

Provide the output of `ls /dev/dri/` to confirm.

Try running:

```sh
LIBVA_DRIVER_NAME=i965 vainfo --display drm --device /dev/dri/renderD128
```

And report back.

Igortorrente, 2022-12-10:
Yeah, this is the case. I also have an AMD GPU in my system.

Your command works!
```
LIBVA_DRIVER_NAME=i965 vainfo --display drm --device /dev/dri/renderD128
libva info: VA-API version 1.16.0
libva info: User environment variable requested driver 'i965'
libva info: Trying to open /usr/lib/x86_64-linux-gnu/dri/i965_drv_video.so
libva info: Found init function __vaDriverInit_1_8
libva info: va_openDriver() returns 0
vainfo: VA-API version: 1.16 (libva 2.12.0)
vainfo: Driver version: Intel i965 driver for Intel(R) Haswell Desktop - 2.4.1
vainfo: Supported profile and entrypoints
      VAProfileMPEG2Simple            :	VAEntrypointVLD
      VAProfileMPEG2Simple            :	VAEntrypointEncSlice
      VAProfileMPEG2Main              :	VAEntrypointVLD
      VAProfileMPEG2Main              :	VAEntrypointEncSlice
      VAProfileH264ConstrainedBaseline:	VAEntrypointVLD
      VAProfileH264ConstrainedBaseline:	VAEntrypointEncSlice
      VAProfileH264Main               :	VAEntrypointVLD
      VAProfileH264Main               :	VAEntrypointEncSlice
      VAProfileH264High               :	VAEntrypointVLD
      VAProfileH264High               :	VAEntrypointEncSlice
      VAProfileH264MultiviewHigh      :	VAEntrypointVLD
      VAProfileH264MultiviewHigh      :	VAEntrypointEncSlice
      VAProfileH264StereoHigh         :	VAEntrypointVLD
      VAProfileH264StereoHigh         :	VAEntrypointEncSlice
      VAProfileVC1Simple              :	VAEntrypointVLD
      VAProfileVC1Main                :	VAEntrypointVLD
      VAProfileVC1Advanced            :	VAEntrypointVLD
      VAProfileNone                   :	VAEntrypointVideoProc
      VAProfileJPEGBaseline           :	VAEntrypointVLD
```

But should be possible to force a specific hardware in the vaapi, right?

Brainiarc7, 2022-12-10:
Globally, no. Only the library driver variable can be set system wide.

Specific apps, such as ffmpeg, can pick select DRM devices and kernel
drivers explicitly at launch.

Igortorrente, 2022-12-10:
Is it possible, in some way, to hide a device for a app that doesn't support explicit DRM device selection?

Brainiarc7, 2022-12-11:
No, not to my knowledge.

A while back, there was a request for this feature in libva but the RFC
never went through. Let me check for that thread.

Brainiarc7, 2022-12-11:
Here it is
https://github.com/intel/libva/issues/221

Igortorrente, 2022-12-11:
> No, not to my knowledge. A while back, there was a request for this feature in libva but the RFC never went through. Let me check for that thread.

Ohh, this is unfortunate :(

> Here it is
[intel/libva#221](https://github.com/intel/libva/issues/221)

 I will close this issue in favor of that one.

closed 2022-12-11 by Igortorrente