intel/intel-vaapi-driver #551: upgrade from 5.18 to 5.19 kernel breaks interlaced video playback

warpme, 2022-08-14:
I'm developer of https://github.com/warpme/minimyth2

Recently I updated kernel from 5.18 to 5.19 and started to have interlaced video playback issue when I'm using LIBVA_DRIVER_NAME=i965. On progressive content playback with i965 driver seems to be ok.

Also using LIBVA_DRIVER_NAME=iHD seems to be also ok.

It looks to me like issue is similar to https://lore.kernel.org/lkml/0249066a-2e95-c21d-d16a-fba08c633c0b@lio96.de/#r.

Is this known issue?

a-andre, 2022-08-17:
The linked issue should be fixed by https://git.kernel.org/pub/scm/linux/kernel/git/stable/linux.git/commit/?id=ad765fae792e16ce3c1d0b69ce939e3f7dba40ab.

@warpme could you bisect to find the kernel commit that causes the issue?

warpme, 2022-08-17:
@a-andre
Yes. I'll try bisect next week!

warpme, 2022-08-23:
pls find bisect results:

```
[piotro@builder-x86-64 linux-5.19-rc3]$ git bisect bad a111daf0c53ae91e71fd2bfe7497862d14132e3e
[piotro@builder-x86-64 linux-5.19-rc3]$ git bisect good b13baccc3850ca8b8cccbf8ed9912dbaa0fdf7f3
Bisecting: 151 revisions left to test after this (roughly 7 steps)
[7c2d03f15f52fb92d3a0602995fd1fb8fbffd475] Merge tag 'linux-watchdog-5.19-rc3' of git://www.linux-watchdog.org/linux-watchdog
[piotro@builder-x86-64 linux-5.19-rc3]$
[piotro@builder-x86-64 linux-5.19]$ git bisect bad
You need to run this command from the toplevel of the working tree.
[piotro@builder-x86-64 linux-5.19]$
[piotro@builder-x86-64 linux-5.19-rc3]$ git bisect bad
Bisecting: 90 revisions left to test after this (roughly 6 steps)
[30306f6194cadcc29c77f6ddcd416a75bf5c0232] Merge tag 'hardening-v5.19-rc3' of git://git.kernel.org/pub/scm/linux/kernel/git/kees/linux
[piotro@builder-x86-64 linux-5.19-rc3]$ git bisect good
Bisecting: 49 revisions left to test after this (roughly 6 steps)
[0639b599f6f3cc871a9e024481a25a7d85946eb8] Merge tag 'audit-pr-20220616' of git://git.kernel.org/pub/scm/linux/kernel/git/pcmoore/audit
[piotro@builder-x86-64 linux-5.19-rc3]$ git bisect good
Bisecting: 19 revisions left to test after this (roughly 5 steps)
[9057a646446c7ff3f1e7a97d0039bde7e7936279] Merge tag 'usb-5.19-rc3' of git://git.kernel.org/pub/scm/linux/kernel/git/gregkh/usb
[piotro@builder-x86-64 linux-5.19-rc3]$ git bisect bad
Bisecting: 14 revisions left to test after this (roughly 4 steps)
[65cf7c02cfc7c36a3e86f3da5818dabc32c270ff] Merge tag 'exynos-drm-fixes-v5.19-rc3' of git://git.kernel.org/pub/scm/linux/kernel/git/daeinki/drm-exynos into drm-fixes
[piotro@builder-x86-64 linux-5.19-rc3]$ git bisect bad
Bisecting: 5 revisions left to test after this (roughly 3 steps)
[3f0acf259a287b16ec0af8ea8b2ee3fb37575d92] Merge tag 'drm-intel-fixes-2022-06-16' of git://anongit.freedesktop.org/drm/drm-intel into drm-fixes
[piotro@builder-x86-64 linux-5.19-rc3]$ git bisect bad
Bisecting: 4 revisions left to test after this (roughly 2 steps)
[2636e008112465ca54559ac4898da5a2515e118a] drm/i915/uc: remove accidental static from a local variable
[piotro@builder-x86-64 linux-5.19-rc3]$ git bisect bad
Bisecting: 1 revision left to test after this (roughly 1 step)
[6e3f3c239ee547c5b55a85f467c92a6ba7eee83a] drm/i915/gt: Fix memory leaks in per-gt sysfs
[piotro@builder-x86-64 linux-5.19-rc3]$ git bisect good
Bisecting: 0 revisions left to test after this (roughly 0 steps)
[842d9346b2fdda4d2fb8ccb5b87faef1ac01ab51] drm/i915: Individualize fences before adding to dma_resv obj
[piotro@builder-x86-64 linux-5.19-rc3]$ git bisect bad
842d9346b2fdda4d2fb8ccb5b87faef1ac01ab51 is the first bad commit
commit 842d9346b2fdda4d2fb8ccb5b87faef1ac01ab51
Author: Nirmoy Das <nirmoy.das@intel.com>
Date:   Wed May 25 11:59:55 2022 +0200

    drm/i915: Individualize fences before adding to dma_resv obj

    _i915_vma_move_to_active() can receive > 1 fences for
    multiple batch buffers submission. Because dma_resv_add_fence()
    can only accept one fence at a time, change _i915_vma_move_to_active()
    to be aware of multiple fences so that it can add individual
    fences to the dma resv object.

    v6: fix multi-line comment.
    v5: remove double fence reservation for batch VMAs.
    v4: Reserve fences for composite_fence on multi-batch contexts and
        also reserve fence slots to composite_fence for each VMAs.
    v3: dma_resv_reserve_fences is not cumulative so pass num_fences.
    v2: make sure to reserve enough fence slots before adding.

    Closes: https://gitlab.freedesktop.org/drm/intel/-/issues/5614
    Fixes: 544460c33821 ("drm/i915: Multi-BB execbuf")
    Cc: <stable@vger.kernel.org> # v5.16+
    Signed-off-by: Nirmoy Das <nirmoy.das@intel.com>
    Reviewed-by: Matthew Auld <matthew.auld@intel.com>
    Reviewed-by: Andrzej Hajda <andrzej.hajda@intel.com>
    Signed-off-by: Matthew Auld <matthew.auld@intel.com>
    Link: https://patchwork.freedesktop.org/patch/msgid/20220525095955.15371-1-nirmoy.das@intel.com
    (cherry picked from commit 420a07b841d03f6a436d8c06571c69aa5c783897)
    Signed-off-by: Jani Nikula <jani.nikula@intel.com>

 drivers/gpu/drm/i915/gem/i915_gem_execbuffer.c |  3 +-
 drivers/gpu/drm/i915/i915_vma.c                | 48 +++++++++++++++-----------
 2 files changed, 30 insertions(+), 21 deletions(-)
[piotro@builder-x86-64 linux-5.19-rc3]$
```

paulmenzel, 2022-09-07:
I created issue [*\[Regression v5.19-rc3\] Broken interlaced video playback*](https://gitlab.freedesktop.org/drm/intel/-/issues/6749) in the Linux drm/intel issue tracker.

nirmoy, 2022-09-07:
Thanks @paulmenzel 
@warpme Could you please apply  https://patchwork.freedesktop.org/patch/501819/?series=108258&rev=1 and let me know if that helps?

warpme, 2022-09-09:
@nirmoy 
I applied https://patchwork.freedesktop.org/patch/501819/?series=108258&rev=1 and this HELPS.
With this patch, 5.19.7 and i965 libva works ok.

nirmoy, 2022-09-09:
@warpme  Thanks for you confirmation. We have merged it intel-gfx the fix should trickle down to other stable versions as well.

warpme, 2022-09-10:
Many thx!

closed 2022-09-10 by warpme