intel/intel-vaapi-driver #554: Upgrade from 5.18 to 5.19 kernel breaks VAAPI video playback

rg4github, 2022-09-17:
Greetings,

Following a kernel upgrade from 5.18.19-100 to 5.19.6-100, VAAPI hardware playback stopped working, showing massively garbled video that often ends in a green screen.

I have since tried kernel versions 5.19.7-100 and 5.19.8-100, as well as the patch for issue #551 ("upgrade from 5.18 to 5.19 kernel breaks interlaced video playback"), but none of those solved the problem.

Sample kernel errors:

Sep 17 12:45:05 tux kernel: i915 0000:00:02.0: [drm] Resetting vecs0 for preemption time out
Sep 17 12:45:05 tux kernel: i915 0000:00:02.0: [drm] mythfrontend[1048] context reset due to GPU hang
Sep 17 12:45:05 tux kernel: i915 0000:00:02.0: [drm] GPU HANG: ecode 9:8:cff8ffef, in mythfrontend [1048]
Sep 17 12:45:06 tux kernel: i915 0000:00:02.0: [drm] Resetting vcs0 for CS error
Sep 17 12:45:06 tux kernel: i915 0000:00:02.0: [drm] mythfrontend[1048] context reset due to GPU hang
Sep 17 12:45:06 tux kernel: i915 0000:00:02.0: [drm] GPU HANG: ecode 9:4:cbfcffe7, in mythfrontend [1048]
Sep 17 12:46:06 tux kernel: Asynchronous wait on fence 0000:00:02.0:mythfrontend[1048]:a7c timed out (hint:intel_atomic_commit_ready [i915])
Sep 17 12:46:07 tux kernel: i915 0000:00:02.0: [drm] Resetting vecs0 for preemption time out

(This was with the stock Fedora 35 driver for 5.19.8-100.)

Hardware info:

smpboot: CPU0: Intel(R) Celeron(R) J4105 CPU @ 1.50GHz (family: 0x6, model: 0x7a, stepping: 0x1)

00:00.0 Host bridge: Intel Corporation Gemini Lake Host Bridge (rev 03)
00:00.1 Signal processing controller: Intel Corporation Celeron/Pentium Silver Processor Dynamic Platform and Thermal Framework Processor Participant (rev 03)
00:02.0 VGA compatible controller: Intel Corporation GeminiLake [UHD Graphics 600] (rev 03)
00:0e.0 Audio device: Intel Corporation Celeron/Pentium Silver Processor High Definition Audio (rev 03)
00:0f.0 Communication controller: Intel Corporation Celeron/Pentium Silver Processor Trusted Execution Engine Interface (rev 03)
00:12.0 SATA controller: Intel Corporation Celeron/Pentium Silver Processor SATA Controller (rev 03)
00:13.0 PCI bridge: Intel Corporation Gemini Lake PCI Express Root Port (rev f3)
00:13.1 PCI bridge: Intel Corporation Gemini Lake PCI Express Root Port (rev f3)
00:13.2 PCI bridge: Intel Corporation Gemini Lake PCI Express Root Port (rev f3)
00:13.3 PCI bridge: Intel Corporation Gemini Lake PCI Express Root Port (rev f3)
00:15.0 USB controller: Intel Corporation Celeron/Pentium Silver Processor USB 3.0 xHCI Controller (rev 03)
00:1f.0 ISA bridge: Intel Corporation Celeron/Pentium Silver Processor LPC Controller (rev 03)
00:1f.1 SMBus: Intel Corporation Celeron/Pentium Silver Processor Gaussian Mixture Model (rev 03)
03:00.0 Ethernet controller: Realtek Semiconductor Co., Ltd. RTL8111/8168/8411 PCI Express Gigabit Ethernet Controller (rev 15)
04:00.0 SATA controller: ASMedia Technology Inc. ASM1062 Serial ATA Controller (rev 02)

I'd be happy to provide additional info.

Kind regards,

- Richard.

praxis88, 2022-09-24:
I found this today looking around for an explanation of what I was seeing. I went to the package repos then followed upstream until I landed here and went into the issues

I am seeing this using a braswell with i965, the video is very garbled, very pixelated, I only saw the green screen once though.

rg4github, 2022-09-24:
> I am seeing this using a braswell with i965, the video is very garbled, very pixelated, I only saw the green screen once though.

Thanks for chiming in. Do you get similar errors in /var/log/messages as well? ("Resetting vecs0 for preemption time out", "GPU Hang", "Asynchronous wait on fence", etc.)

praxis88, 2022-09-24:
I don't see anything like that in my logs. I do see the browser throwing a bunch of errors about vsync when attempting to use hw decode. Then the entire desktop locks up for about 20 seconds while it attempts to make an image

mpv -hwdec=vaapi file 

Plays the file, doesn't throw any errors, but the video output it makes is a hot mess in a way it wasn't before.

praxis88, 2022-10-03:
Braswell only works at all with i965

Broadwell and everything later is ihd.

Ihd wont even intialize on my machine.

praxis88, 2022-10-08:
Whatever broke this, this is no longer broken for me in 5.19.13-arch1-1

rg4github, 2022-10-09:
I can confirm the issue is no longer broken for me either after upgrading to 5.19.13-100 (Fedora 35).

The same system update also included new Intel GPU firmware (intel-gpu-firmware-20220815-139.fc35.noarch), but reverting back to kernel 5.19.10-100 reintroduce the issue, so the problem does appear to have been fixed by the kernel+kmod combo.

Closing this issue as resolved.

closed 2022-10-09 by rg4github