intel/intel-vaapi-driver #563: How to check if HW decoding is supported for model "X"?

Talkless, 2023-09-13:
Hi,

Is there a way to check in code (or elsewhere) if HW decoding is implemented for one or another particular CPU model?

For example, currently I am curious if h264 hardware decoding should™ work on Intel Atom x5-Z8350 using va-api.

Thanks.

BigCojones, 2024-04-23:
Wow, nobody answered? :smile:

First you would want to know the exact architecture for your chip or the "Code name". 
Found example here:
https://ark.intel.com/content/www/us/en/ark/products/93361/intel-atom-x5-z8350-processor-2m-cache-up-to-1-92-ghz.html

So now we know it's a "Cherry trail" chip. 

Then you would search for something along the lines of Cherry trail and video decoding.
My search got me this wikipedia chart:
https://en.wikipedia.org/wiki/Intel_Quick_Sync_Video#Hardware_decoding_and_encoding

Which shows Cherry trail having full support for AVC (another name for h.264) on quick sync.

So I would take that as a guarantee that it should work pretty flawlessly. And if there is any problem with the decoding, THEN you would make an new issue here and explain the problem. It should work according to spec.

Talkless, 2024-04-26:
It's not the question if chip supports HW decoding, but if this library is updated to support that chip (and codec), and at witch version if so.

uartie, 2024-04-26:
You can check if your pciid is listed in https://github.com/intel/intel-vaapi-driver/blob/master/src/i965_pciids.h.  If not, you can try to add it... but there is no guarantees it will work.

You may also run `vainfo` (provided by https://github.com/intel/libva-utils) to see what the driver capabilities are.

Either way, since that HW is obsolete and this driver is no longer being developed, there is no guarantees or support provided.

Talkless, 2024-04-26:
> You can check if your pciid is listed in

Thansk, that's useful!

> You may also run vainfo

That's too late. I need to figure out BEFORE buying the chip. Like, "will this cheap chinese fanless pc with integrated LCD panel using Intel XYZ CPU will handle this or that codec on Debian 12 with libva version x".