intel/intel-vaapi-driver #564: [BUG] Problems with VaapiVideoDecodeGL from Chromium in Wayland

EliasOfWaffle, 2023-10-13:
Chromium have a new VAAPI decoder for OpenGL renderer, it supports Wayland and can run great with new intel generations using media driver and onevpl driver, but for the i965 is the contrary have some reports in arch forum that i965 don't work in some hardware that can support intel media driver and after switch for intel media driver it's work.
But for older generations switch is not possible, including in Gen7 the bigger problem is:
             ````````
  vaapi video decoder.cc(1245)] : failed initializing the frame pool
             ````````

EliasOfWaffle, 2023-10-13:
if running good in others drivers, can be a specific for i965 or is a chromium bug?

PD-Kerman, 2024-08-03:
> Chromium have a new VAAPI decoder for OpenGL renderer, it supports Wayland and can run great with new intel generations using media driver and onevpl driver, but for the i965 is the contrary have some reports in arch forum that i965 don't work in some hardware that can support intel media driver and after switch for intel media driver it's work.Chromium 有一个用于 OpenGL 渲染器的新 VAAPI 解码器，它支持 Wayland，并且可以使用媒体驱动程序和 onevpl 驱动程序在新一代英特尔上运行良好，但对于 i965 则相反，在 arch 论坛中有一些报告称 i965 无法在某些硬件中工作可以支持英特尔媒体驱动程序，切换为英特尔媒体驱动程序后即可工作。 But for older generations switch is not possible, including in Gen7 the bigger problem is:但对于老一代来说，切换是不可能的，包括 Gen7，更大的问题是： `vaapi video decoder.cc(1245)] : failed initializing the frame pool`

is there any workarounds now~? 
i got the same issue when trying to get hardware-acceleration on a N3160(BSW) Gen8 HD400 machine, and i got exactly the same error as showned
`vaapi video decoder.cc(1245)] : failed initializing the frame pool`
when i manually add the 
`--enable-features=VaapiVideoDecoder` flag in the command line and verified in the chrome://version page
and thats basically how i got the error and its also shown in `chrome://media-internal`
firefox kinda work i can see some activities shown in the `intel_gpu_top` interface and ive been stuck on this issue for like a few weeks and nomatter how i try i just cant get any video decoding activities with chrome/chromium based browsers i.e. no hw-accel for video streaming
vainfo were checked and its all fine as well
and i think jve got some similar issue on a 8350u machine with uhd620 igpu running debian with xfce,
i didnt specify the LIBVA_DRIVER_NAME yet tho cuz it seems to be working just fine when printing vainfos *it directly chosed the i965 one

colaas, 2024-12-22:
@PD-Kerman did you find a fix?
I have the same Braswell GPU and vaapi works fine with mpv (libva-intel-driver - Intel i965 driver for Intel(R) CherryView - 2.4.1) but after trying hundreds of possible combinations there is no way I found to make it work on chromium it always breaks with:
`
 vaapi video decoder.cc(1245)] : failed initializing the frame pool`