intel/intel-vaapi-driver #565: [Bug]: VAAPI HEVC decoding hangs GPU

0anton, 2024-02-06:
vaapi driver stops working with _GPU hang_ messages decoding HENC video stream from Reolink 811A camera.

Hardware:
- Intel Jasperlake (Gen11) @ /dev/dri/card0

Kernel:
- Linux 6.1.0-13-amd64 #1 SMP PREEMPT_DYNAMIC Debian 6.1.55-1 (2023-09-29) x86_64 GNU/Linux

dmesg:
```
[1479007.361603] i915 0000:00:02.0: [drm] ffmpeg[396018] context reset due to GPU hang
[1479007.363887] i915 0000:00:02.0: [drm] GPU HANG: ecode 11:4:a8fffffd, in ffmpeg [396018]
[1489435.651010] i915 0000:00:02.0: [drm] Resetting vcs0 for preemption time out
[1489435.651724] i915 0000:00:02.0: [drm] *ERROR* vcs0 reset request timed out: {request: 00000001, RESET_CTL: 00000001}
[1489435.657380] i915 0000:00:02.0: [drm] GPU HANG: ecode 11:4:486dffef, in ffmpeg [405354]
[1489450.536887] i915 0000:00:02.0: [drm] GPU HANG: ecode 11:4:486dffef, in ffmpeg [405354]
[1489450.538002] i915 0000:00:02.0: [drm] Resetting vcs0 for stopped heartbeat on vcs0
[1489450.538709] i915 0000:00:02.0: [drm] *ERROR* vcs0 reset request timed out: {request: 00000001, RESET_CTL: 00000001}
[1489450.538880] i915 0000:00:02.0: [drm] Resetting chip for stopped heartbeat on vcs0
[1489450.641813] i915 0000:00:02.0: [drm] *ERROR* vcs0 reset request timed out: {request: 00000001, RESET_CTL: 00000001}
[1489450.642547] i915 0000:00:02.0: [drm] *ERROR* vcs0 reset request timed out: {request: 00000001, RESET_CTL: 00000001}
[1489450.645345] i915 0000:00:02.0: [drm] *ERROR* vcs0 reset request timed out: {request: 00000001, RESET_CTL: 00000001}
[1489450.665793] i915 0000:00:02.0: [drm] *ERROR* vcs0 reset request timed out: {request: 00000001, RESET_CTL: 00000001}
[1489450.666513] i915 0000:00:02.0: [drm] *ERROR* vcs0 reset request timed out: {request: 00000001, RESET_CTL: 00000001}
[1489450.669286] i915 0000:00:02.0: [drm] *ERROR* vcs0 reset request timed out: {request: 00000001, RESET_CTL: 00000001}
[1489450.697789] i915 0000:00:02.0: [drm] *ERROR* vcs0 reset request timed out: {request: 00000001, RESET_CTL: 00000001}
[1489450.698525] i915 0000:00:02.0: [drm] *ERROR* vcs0 reset request timed out: {request: 00000001, RESET_CTL: 00000001}
[1489450.701326] i915 0000:00:02.0: [drm] *ERROR* vcs0 reset request timed out: {request: 00000001, RESET_CTL: 00000001}
[1489450.741792] i915 0000:00:02.0: [drm] *ERROR* vcs0 reset request timed out: {request: 00000001, RESET_CTL: 00000001}
[1489450.742513] i915 0000:00:02.0: [drm] *ERROR* vcs0 reset request timed out: {request: 00000001, RESET_CTL: 00000001}
[1489450.745285] i915 0000:00:02.0: [drm] *ERROR* vcs0 reset request timed out: {request: 00000001, RESET_CTL: 00000001}
[1489450.747346] i915 0000:00:02.0: [drm] *ERROR* Failed to reset chip
[1489450.747349] i915 0000:00:02.0: [drm:add_taint_for_CI [i915]] CI tainted:0x9 by intel_gt_reset+0x23a/0x2a0 [i915]
[1489450.749520] [drm:__uc_sanitize [i915]] *ERROR* Failed to reset GuC, ret = -110
[1489450.852369] i915 0000:00:02.0: [drm] *ERROR* vcs0 reset request timed out: {request: 00000001, RESET_CTL: 00000001}
[1489450.853102] i915 0000:00:02.0: [drm] *ERROR* vcs0 reset request timed out: {request: 00000001, RESET_CTL: 00000001}
[1489450.855903] i915 0000:00:02.0: [drm] *ERROR* vcs0 reset request timed out: {request: 00000001, RESET_CTL: 00000001}
[1489450.858048] i915 0000:00:02.0: [drm] ffmpeg[405354] context reset due to GPU hang
```

GPU:
```
00:02.0 VGA compatible controller: Intel Corporation JasperLake [UHD Graphics] (rev 01)
        DeviceName: Onboard - Video
        Subsystem: Intel Corporation JasperLake [UHD Graphics]
        Kernel driver in use: i915
        Kernel modules: i915
```

ffmpeg command:
```
ffmpeg -hide_banner -loglevel warning -threads 2 -hwaccel_flags allow_profile_mismatch -hwaccel vaapi -hwaccel_device /dev/dri/renderD128 -hwaccel_output_format vaapi -user_agent FFmpeg Frigate/0.13.0-49814b3 -rtsp_transport tcp -timeout 5000000 -i rtsp://localhost:8554/camera2 -f segment -segment_time 10 -segment_format mp4 -reset_timestamps 1 -strftime 1 -c copy
```

ffprobe output:
```
Stream 0:
Return Code: 0

Video:

Codec: H.265 / HEVC (High Efficiency Video Coding)
Resolution: 3840x2160
FPS: Unknown

Audio:

Codec: AAC (Advanced Audio Coding)
```

eero-t, 2024-03-18:
Please use the correct driver: https://github.com/intel/media-driver/blob/master/README.md#supported-platforms

And close this ticket, as this project is the legacy driver for the legacy HW.

0anton, 2024-03-18:
I don’t quite understand your answer. `JSL (Jasper Lake)` is in the list of supported platforms on the link you provided. Which driver is then recommended for this platform?

eero-t, 2024-03-19:
> I don’t quite understand your answer. `JSL (Jasper Lake)` is in the list of supported platforms on the link you provided. 

For the *intel-media* driver.   This project is *intel-vaapi* driver, a different, legacy VA-API backend for much older HW.

> Which driver is then recommended for this platform?

[https://github.com/intel/media-driver/](https://github.com/intel/media-driver/)

0anton, 2024-03-19:
Ah, thanks for pointing out to this. I’m closing this then as requested!

closed 2024-03-19 by 0anton